Stephane-D/SGDK #173: VDP_setHorizontalScroll

izzidor, 2020-10-03:
Bonjour,
je voudrais faire défiler le plan B de droite à gauche, j'ai dessiné sur le plan B une image BMP de 320*224 avec  la fonction  " VDP_drawImageEx ". J'ai cru comprendre que avec la fonction " VDP_setHorizontalScroll " les pixels en valeur négative étaient automatiquement redéssinés à droite pour un defilement de droite à gauche ou inversement. 

Dans mon cas, effectivement, j'ai bien un décalage des pixels de la droite vers la gauche mais les pixels négative ne sont pas redesinés à droite.
Mon code  : 
**int main()
{
    const int scrollspeed = 2;
    VDP_loadTileSet(back_G.tileset,1,DMA);
    VDP_setPalette(PAL1, back_G.palette->data);

    //affichage background_B
    VDP_drawImageEx(BG_B, &back_G, TILE_ATTR_FULL(PAL1, 0, 0, 0, 1), 0, 0, 0, CPU);
    VDP_setScrollingMode(HSCROLL_PLANE,VSCROLL_PLANE);
    
    int offset = 0;
    while(1)
    {
         VDP_setHorizontalScroll(BG_B,  offset = -scrollspeed);
         if(offset > 320)
         {
             offset=0;
         }

    VDP_waitVSync();
    }
    return (0);
}**

Merci pour l'aide

izzidor, 2020-10-03:
Trouvé, "offset = -scrollspeed  " offset n'était pas incrémenté !!
corrigé avec :  const int scrollspeed = -2   offset =+ scrollspeed

Stephane-D, 2020-10-05:
Salut izzidor,

Les issues c'est vraiment pour remonter des bugs dans SGDK. Pour avoir de l'assistance sur SGDK je te conseille plutot de passer sur le serveur Discord :)
https://discord.gg/xmnBWQS

Ou sur le forum SpritesMind :
http://gendev.spritesmind.net/forum/

closed 2020-10-05 by Stephane-D