Stephane-D/SGDK #178: SGDK can not be compiled on modern non-Windows machines using GCC 6

bferguson3, 2020-11-14:
GCC 6 is old and has too many issues at this point to successfully compile on anything other than an x86-32 compatible Windows machine. 

I've attempted both the Mac and Linux gendev branches to no avail. I also tried compiling on the new Raspberry Pi 400. I've encountered a variety of errors on both, the most notable are `Specified CC_FOR_BUILD doesn't seem to work` on MacOS and something similar to `-1 cannot be coerced to null` on raspbian. 

32-bit dependencies must be removed moving forward - ARM64 is picking up steam, and right now the only way to make Genesis games with this SDK is on Windows 10. Seems odd, no?

GCC 7, 8 and 9 are all maintained still, whereas 6 is not. Please abandon it? Please?

Stephane-D, 2020-11-16:
@bferguson3 Nothing prevent to use newer GCC. I embed GCC 6.3 binaries for Windows platform for convenience but you can probably modify / patch [GenDev ](https://github.com/kubilus1/gendev) project so you can build newer GCC without any issues. I know that somepeople use SGDK with newer GCC builds (as GCC 9).

matteusbeus, 2020-11-16:
Does it have any effect on the resulting binary performance wise?

Stephane-D, 2020-11-16:
GCC >= 6 are about the same level of performance. From different tests it looks like GCC 6.x still generate a better code for 68k target but the difference isn't important

Stephane-D, 2021-07-08:
Can be closed (any recent GCC version can be used).

closed 2021-07-08 by Stephane-D