Stephane-D/SGDK #179: Error on sprintf visual studio

trialforce, 2020-12-05:
Hi!

I'm not a a C developer but I'm get  "spected {" on string.h line 159.

Is the sprintf definition.

u16 sprintf(char *buffer,const char *fmt, ...) __attribute__ ((format (printf, 2, 3)));

If I comment this line I can compile my rom. 

How can I solve it?

Stephane-D, 2020-12-07:
Hi,

This is to enforce GCC checking parameters provided in a specific way for this function (correct type provided from the input string format) :
https://gcc.gnu.org/onlinedocs/gcc-3.2/gcc/Function-Attributes.html

This is a GCC stuff and normally if you properly installed SGDK with included GCC binaries it shouldn't complain about it.
How did you configured your project in Visual Studio ? i guess the problem is there..

trialforce, 2020-12-07:
Hi Stephane!

Thanks for your great work on SGDK and for your anwser.

Sorry, I don't understad very much about C and GCC, I unzip the downloaded SGDK package and put in c:/sgdk.
I put some paths on my windows and im Visual studio I created a makefile project with this command line:

%GDK_WIN%\bin\make -f %GDK_WIN%\makefile.gen

I only did that.

Do you have some link that explain the proccess?

Stephane-D, 2020-12-07:
Did you added the bin folder of SGDK into your _PATH_ environment variable ? this is important to get SGDK GCC (C compiler) working correctly.
To be honest without good knowledge in C, developing with SGDK will be difficult as it implies you already have an experience with C.

trialforce, 2020-12-07:
Thanks again Stephane!

I included the BIN and it starts working. The tutorial I watched only said about "inc" folder.
I'm a programmer for almost 15 years (python, java, android, c#, php, javascript),
I already Know C, but I'm not that good.
By tha way, I'm already getting good results with SGDK, thanks.

closed 2020-12-07 by trialforce

Stephane-D, 2020-12-08:
Unfortunately there is lot of wrong tutorials online :-/
You can just check into SGDK wiki to get proper instructions to start at least.
SGDK Wiki definitely contains not much, but you should have basics to start, then you can follow more advanced tutorials :)