Stephane-D/SGDK #180: VDP_drawText > x pos overflows VRAM

bferguson3, 2020-12-11:
Steps
1. Make any example project
2. Add code: VDP_drawText("String", 60, 0);
3. Observe rom crashes

This appears to be arbitrary, as the Y offset doesn't crash. This should either be marked "unsafe" (eg VDP_drawTextUnsafe) or fixed so that it won't cause this headache.

Stephane-D, 2020-12-11:
I can make these methods specifically more safe but in general adding that kind of safety does make SGDK a bit slower as we are running C code on a 7.7 Mhz 68000. SGDK does still contains some barriers here and there but there is always trade-off between performance and safety so unfortunately you can't expect every methods to be inherently safe.

Stephane-D, 2020-12-11:
I specifically protected VDP_drawText() / VDP_clearText() methods but to be honest i'm not really sure that is a good idea as a lot of other methods (as VDP_fillTileMapRect(..) for instance) aren't safe and i don't think it would be a good idea to try to make all these methods safe for global performance.

closed 2020-12-11 by Stephane-D

bferguson3, 2020-12-11:
Please note that the way this routine *appears* is coded, it requires the tile offset within the MAP, not VRAM.

bferguson3, 2020-12-11:
I understand, I was getting unpredictable results because of my tilemap
size. I think narrowing the map to size of BG planes helps a lot.