Stephane-D/SGDK #181: Minimum res tilemap/map displays vram garbage

bferguson3, 2020-12-11:
1. Create two images, 320x224
2. Create new project with new TILEMAP, PALETTE, MAP from single image
3. Initialize to set BG A to image 1, BG B to image 2
3. MAP_scrollTo(mapa, 0,0);
MAP_scrollTo(mapb, 0,0);
4. Observe vram garbage (especially obvious with plane debugger)

Stephane-D, 2020-12-11:
Can you post your complete project (resources and sources files) ?
I don't understand if you have a single image for both MAP or not, if you have 2 images my first guess here is that you turn out of VRAM but without details i can't figure that..

bferguson3, 2020-12-11:
A minimal project won't reproduce the issue, so I am assuming there is some issue with my combination of res files

closed 2020-12-11 by bferguson3

bferguson3, 2020-12-11:
Figured out my issue here, it was misuse of VDP_drawText. I was performing the operation before SYS_doVblankProcess().

Stephane-D, 2020-12-14:
Hmm.. weird that shouldn't be a issue honestly (in worst case it would appear on next frame). Still glad to hear you fixed it :)