Stephane-D/SGDK #182: Environment path issues; string.h cannot be included

bferguson3, 2020-12-11:
1. The makecart bat contains white space and does not include C: in the linux-style path, which causes it to fail to run, obviously.
Bolded:
set GDK=**c:**/sgdk**_**
GDK_WIN=C:\sgdk**_**
PATH=**c:**/sgdk;C:\sgdk;C:\sgdk\bin
C:\sgdk\bin\make -f C:\sgdk\makefile.gen

2. Windows will never include string.h because of the way it is defined in the header.
```
#if (ENABLE_NEWLIB == 1) && !defined(_NEWLIB_STRING_H_)
#define _NEWLIB_STRING_H_
#include_next <string.h> // Include string.h from newlib
#undef _STRING_H_        // Will be defined again just below
#endif

#ifndef _STRING_H_
#define _STRING_H_
```
I don't know what newlib is, but regardless, this is always ignored by the compiler with the options you have set up by default. Please migrate this code to something like "genstring.h" to avoid silly conflicts.

Stephane-D, 2020-12-11:
Thanks for reporting !
I just fixed 1. (the batch wasn't from me but i should have tested it, i guess it worked on older OS).

For point 2. : NEWLIB is a replacement to default GLIBC, to be honest SGDK is not intended to be used with Newlib but some people want to use it so they can modify the config.h file to enable it, in which case some newlib methods will cohabit with defaults SGDK ones. And SGDK does use _STRING_H_ as that should be the *default String* implementation, newlib is actually the alternative one in that case.

closed 2020-12-11 by Stephane-D

bferguson3, 2020-12-11:
Let me try rephrasing my problem:

fix16toStr() is undefined

Stephane-D, 2020-12-11:
It is (just tried to be sure)... you probably messed up something.

bferguson3, 2020-12-11:
```c:/sgdk/bin/gcc -nostdinc -m68000 -Wall -Wextra -Wno-shift-negative-value -Wno-main -Wno-unused-parameter -fno-builtin -Iinc -Isrc -Ires -Ic:/sgdk/inc -Ic:/sgdk/res -Bc:/sgdk/bin -O3 -fuse-linker-plugin -fno-web -fno-gcse -fno-unit-at-a-time -fomit-frame-pointer -flto -MMD -c src/main.c -o out/src/main.o
src/main.c: In function 'main':
src/main.c:179:21: warning: implicit declaration of function 'fix16toStr' [-Wimplicit-function-declaration]
         char* var = fix16toStr(ticker);
                     ^~~~~~~~~~```
It insists on including the built-in string.h first. Do you have other build environments on your machine? Why does your include path not find those? Are you building with a visual studio project and accompanying settings, or a blank project and command line batch?

bferguson3, 2020-12-11:
Also, your toolchain requires java to be on the path if using any res files. 

The batch should be:
```
set GDK=C:/sgdk
GDK_WIN=C:\sgdk;
PATH=C:\sgdk;C:\sgdk\bin;C:\Program Files\Java\jdk1.8.0_271\bin
C:\sgdk\bin\make -f C:\sgdk\makefile.gen
```

Stephane-D, 2020-12-11:
I don't have other build environment and I indeed simply use the default makefile to build up my project (i use Code::Blocks IDE which can build from makefile) :
If you go to wiki, SGDK usage page:  https://github.com/Stephane-D/SGDK/wiki/SGDK-Usage

You can see you simply need to execute that command to build up your project:
`%GDK_WIN%\bin\make -f %GDK_WIN%\makefile.gen`

And here is an example i just tried with a fix16ToStr(..) call :
[image]

So i guess there is something not properly setup in your development environment.

bferguson3, 2020-12-11:
`%GDK_WIN%\bin\make -f %GDK_WIN%\makefile.gen`
This is the command I've been using all this time until trying to debug this issue.
I'll try Code::Blocks, perhaps you have something odd going on there with paths.

Obviously, I've seen the wiki. I wouldn't be this far if I haven't.

Stephane-D, 2020-12-11:
> 
> 
> Also, your toolchain requires java to be on the path if using any res files.
> 
> The batch should be:
> 
> ```
> set GDK=C:/sgdk
> GDK_WIN=C:\sgdk;
> PATH=C:\sgdk;C:\sgdk\bin;C:\Program Files\Java\jdk1.8.0_271\bin
> C:\sgdk\bin\make -f C:\sgdk\makefile.gen
> ```

To be honest the script was done a lot time ago when java wasn't yet used, that explains the problem.
Still we can expect the java bin to already be in path if properly installed (we cannot easily find where java is installed and which version).

> Obviously, I've seen the wiki. I wouldn't be this far if I haven't.

Then i can just repeat you there is something wrong somewhere with your dev end as there is nothing involded in code::blocks here, i just shown you the result executing _from command line_
Is only fix16ToStr(..) not found ? or any string methods ? others methods are properly found ? just including "genesis.h" is enough to give you access to full SGDK API.

bferguson3, 2020-12-11:
```

C:\Users\Bent\Documents\Projects\genesis\stringtest>C:\sgdk\bin\make -f C:\sgdk\makefile.gen 
C:/sgdk/bin/mkdir -p src/boot
C:/sgdk/bin/mkdir -p out
C:/sgdk/bin/gcc -nostdinc -BC:/sgdk/bin -n -T C:/sgdk/md.ld -nostdlib out/sega.o @out/cmd_ C:/sgdk/lib/libmd.a C:/sgdk/lib/libgcc.a -o out/rom.out -Wl,--gc-sections
C:\Users\Bent\AppData\Local\Temp\ccko7DaU.ltrans1.ltrans.o: In function `_reset_entry':
<artificial>:(.text+0x1a4): undefined reference to `fix16toStr'
C:\Users\Bent\AppData\Local\Temp\ccko7DaU.ltrans1.ltrans.o: In function `_start_entry':
<artificial>:(.text+0x976): undefined reference to `fix16toStr'
make: *** [out/rom.out] Error 1
```
from this project:
```
#include <genesis.h>
int main(u16 hard) // hard == 0 if reset button was pressed
{
    u16 test = 0;
    while(TRUE)
    {     
        char* var = fix16toStr(test);
        SYS_doVBlankProcess();
    }
    return 0;
}
```
There actually is a small problem with code blocks, when you set up the compiler, you set an execution directory. According to your directions (even though they are out of date, the new code blocks is different) the execution directory is the SGDK directory, but that is not the case for how it works right now. 
(Either way I couldn't get code blocks to compile.)

Stephane-D, 2020-12-11:
I should have look more carefully to your initial message.
The method name is `fix16ToStr(..)` (case is important) then your example doesn't use it correctly (thre is 3 parameters for this method). You have a doxygen describing all the methods in SGDK _doc_ folder, but you can also directly look onto method declaration in .h unit as doxygen comes from there. Depending how you setup your IDE you can have auto-completion and even the doxygen displaying when you're coding, making life much easier when you have to deal with SGDK API.

I guess i can close this one ?

bferguson3, 2020-12-11:
Ugh, thank you!

Gonna shoot myself later.