Stephane-D/SGDK #188: SGDK's compiler toolchain doesn't use prefixed names

ketsuban, 2021-01-27:
Step into the `bin` directory of this very repository and you'll find `ar.exe`, `as.exe`, `gcc.exe`... these names should be reserved for the *system* compiler (supplied by something like MSYS2 or Cygwin if you choose to install it - Windows doesn't ship with a GNU toolchain) while SGDK's toolchain wears a `m68k-elf-`. (Compare for instance devkitARM, which correctly prefixes its toolchain with `arm-none-eabi-`.)

Stephane-D, 2021-02-07:
SGDK use a custom / partial build of GCC and unfortunately i've some issues when trying to use correct name as i guess GCC expect a specific folder structure that we don't have here.

Stephane-D, 2021-03-15:
Cannot be done without adding alias and others stuffs needed by classic GCC installation.
Though **I removed the need of adding SGDK bin folder to the PATH environment variable so that is not anymore an issue** :)
SGDK by itself can now fully work from its installation folder without any need of patching PATH variable or requiring extra environement variable (since SGDK 1.62).

closed 2021-03-15 by Stephane-D