Stephane-D/SGDK #190: MAP_getTile issue

edfurby, 2021-01-31:
I am trying to use this call for collision detection so I need to know if a tile exists at a given position. It correctly returns a tile attribute at a given position on the map if a tile exists, but if no tile exists then it still returns the last known tile attribute. It correctly returns 0 initially if no tile is found but the issue occurs after a tile is found. I tried MAP_getMetaTile and get the same behaviour. Hope this helps.

Stephane-D, 2021-02-01:
Why not testing with the map size before calling the method ?
I can add bounds checking to the method but it will add a small CPU overhead each time you call it and you're likely to call it quite a bit per frame so it's why i want to keep it as fast as possible. Another (and better imo) solution is to define a map larger than the usable area filled with solid tiles for _outside areas_  so you will never need to check about map bounds.. map encoding should be efficient enough so it doesn't add to much to the final map size.

Stephane-D, 2021-03-15:
Map bounds checking is meant to be done outside the function as that would slowdown it quite a bit and we really need to keep it fast (possibly many calls per frame)

closed 2021-03-15 by Stephane-D