Stephane-D/SGDK #196: New complete game created with SGDK!

mxfolken, 2021-03-09:
Hello congratulations on your SGDK, I have created a complete game (not demo) of the Thundercats, and I refer to the SGDK:

https://github.com/mxfolken/thundercats_megadrive

https://nova32-entertainment.itch.io/thundercats-for-sega-genesis

I am writing to you, in case you want to include it in the list of games created with SGDK.

**Best regards!**

Stephane-D, 2021-03-09:
Thanks ! Congratulation to you for this achievement of having completing a full game on Sega Mega Drive !
I will add the link to the readme ! As i want to only refer one link per game i think it would be better to reference the github link, it would be nice to see the itch.io link as well from the github readme so people can find it easily too :)

Best regards and thanks for notified me about it !

closed 2021-03-09 by Stephane-D

mxfolken, 2021-03-09:
Many thanks to you, I am very happy to be able to program games with SGDK.

Receive a big hug!