Stephane-D/SGDK #199: Unable to build any sample

pladaria, 2021-03-14:
Hi,

I'm following README instructions to build in linux using docker:

1. `docker build -t sgdk .`
```
    [...]
    Successfully built 60411313eebd
    Successfully tagged sgdk:latest
```
2. `cd sample/basics/hello-world` (README says `sample/sprite` but that example does not exist)
3. `docker run --rm -v $PWD:/src sgdk`
```
/sgdk/bin/mkdir -p src/boot
wine: created the configuration directory '/sgdk/bin/.wineconf'
Could not load wine-gecko. HTML rendering will be disabled.
wine: configuration in '/sgdk/bin/.wineconf' has been updated.
make: *** [/sgdk/makefile_wine.gen:145: pre-build] Error 66
```

Any ideas?

pladaria, 2021-03-14:
btw, I also tried the wine approach locally with the same results

jespicas, 2021-03-15:
I modified docker file to work with a 64 bit Debian and I have other errror
`wine: created the configuration directory '/sgdk/bin/.wineconf'
Could not load wine-gecko. HTML rendering will be disabled.
Could not load wine-gecko. HTML rendering will be disabled.
wine: configuration in '/sgdk/bin/.wineconf' has been updated.
wine: Bad EXE format for Z:\sgdk\bin\sh.exe.
make: *** [/sgdk/makefile_wine.gen:145: pre-build] Error 193`

pladaria, 2021-03-15:
Meanwhile I've configured my dev environment as:

1. Linux host with my editor and tools (using VSCode)
2. VM Running Windows 10 and a shared mount to the SDK folder (which contains my project files)
3. I created a simple make file which sends the compile command to windows

Demo:
https://user-images.githubusercontent.com/579417/111135489-023eac00-857d-11eb-80bd-6b10f8d5d736.mp4

Sources: 
https://github.com/pladaria/SGDK/tree/master/projects

LIZARDRIVE, 2021-03-20:
Hello,
I am pretty sure this is related to SGDK 1.62 not being compatible with generate_wine.sh script. This may be related to changes in makefile, but I am yet to take a look to confim this.
Try purging your sgdk folder (rm -dr to also get rid of .wineconf) and starting all over again using previous released version (in case you don't use anything specific to 1.62).
You may also change JRE version to 14.

mattsputnikdigital, 2021-04-14:
> Meanwhile I've configured my dev environment as:
> 
> 1. Linux host with my sources, editor (VSCode) and tools
> 2. VM Running Windows 10 and a shared mount to the SDK folder (which contains my project files)
> 3. Simple Makefile to send the compile command to the Windows VM (using `VBoxManage guestcontrol run`)
> 
> Demo:
> https://user-images.githubusercontent.com/579417/111135489-023eac00-857d-11eb-80bd-6b10f8d5d736.mp4

Can you put up your Makefile please, I think im going to try something similar with VMWare

pladaria, 2021-04-14:
@mattsputnikdigital sure!

### Makefile
```make
FOLDER=$(shell basename $(CURDIR))
EMULATOR=wine /home/pladaria/projects/genesis/gens-kmod/Gens_KMod_v0.7.3/gens.exe

build:
	@echo "*** Building $(FOLDER)"
	VBoxManage guestcontrol "Windows 10" --username "pedro" run "Z:\\SGDK\\projects\\$(FOLDER)\make.bat"
	$(EMULATOR) 'z:\home\pladaria\projects\genesis\SGDK\projects\test6\out\rom.bin'

clean:
	rm -rf out
	rm -f res/**/*.h
	rm -f res/*.h
	rm -f res/**/*.rs
	rm -f res/*.rs

```
### make.bat
```bat
echo off

rem change working dir
cd /D "%~dp0"

rem print working dir
cd

rem compile
Z:\SGDK\bin\make.exe -f Z:\SGDK\makefile.gen
```

Copied "as is" you should adapt for your needs.

Some notes:
* Executing `make` in my project folder tells windows to execute the `make.bat` file. The bat file was needed to make sure the working directory is correct
* I have a shared folder `/home/pladaria/projects/genesis` (host) mapped to `Z:` (windows VM)
* I run Gens KMod using wine (note the path to the rom passed as argument)

mattsputnikdigital, 2021-04-17:
OK I have got the VMWare on OSX running as with @pladaria's VBox example.

You need to specify the path to the virtual machine and use vmrun

`VM=/Users/USER/Virtual\ Machines.localized/Windows\ 10\ x64.vmwarevm/Windows\ 10\ x64.vmx

vmrun -gu USERNAME -gp PASSWORD runScriptInGuest $(VM) -interactive "" "cmd.exe /k \"Z:\\Sega\\Projects\\Test\\make.bat\" x86"
`

Elazul, 2021-05-26:
Hi! I've posted a solution that is working for me in case you want to give it a try: https://github.com/Stephane-D/SGDK/pull/201#issuecomment-849081998

LIZARDRIVE, 2021-06-02:
Works perfectly, thank you!

Elazul, 2021-06-02:
> Works perfectly, thank you!

You're welcome! Glad this helps!

pladaria, 2021-06-08:
Fixed in https://github.com/Stephane-D/SGDK/pull/216

Thanks @LIZARDRIVE  and @Elazul for your work!

closed 2021-06-08 by pladaria