Stephane-D/SGDK #202: When using xgmtool to convert xgm to vgm, the generated vgm file has no sound.

denjhang, 2021-03-29:
I am very confused. The following is a music clip I wrote using mml2vgm, using 4-channel pcm.

[1.zip](https://github.com/Stephane-D/SGDK/files/6223028/1.zip)

Stephane-D, 2021-03-30:
It's just that XGM to VGM conversion doesn't support 4 PCM channels so maybe it fails.
You can directly use your XGM file with XGMRomBuilder (i tested and it works, the file you sent generated a 4 second music)

closed 2021-03-30 by Stephane-D

denjhang, 2021-03-30:
I use xgmrombuilder and successfully converted xgm to rom, which can play music in the emulator. But there seems to be something wrong with the tone. The frequency of pcm sampling and playback is half a tone higher than the original xgm. This confuses me. (I used this music written by mml2vgm.)