Stephane-D/SGDK #204: Missing libres.h

moon-watcher, 2021-03-30:
https://github.com/Stephane-D/SGDK/blob/5a3d9cc110b785e41d48af705323982c5ade3ac1/inc/font.h#L18

[libres.h not found in inc/ folder again]( http://gendev.spritesmind.net/forum/viewtopic.php?p=22685#p22685)

> ...the file is auto generated when you compile the library so you can try to recompile the library before compiling your project.
> For that you just need to use make but with a different makefile :
> make.exe -f D:\apps\gendev\makelib.gen

sp097326, 2021-03-30:
In my case, I found libres.h in the \\res folder. I pointed my .vscode file to  ${env:GDK}\\res folder additional to the inc folder. Hope this helps.

Stephane-D, 2021-03-30:
Indeed this is an auto generated .h file from rescomp compiler so it's why it ends in the _res_ folder and the SGDK default makefile *does* include .h files from res folder.

closed 2021-03-30 by Stephane-D