Stephane-D/SGDK #205: Question about tilesets/maps

pladaria, 2021-04-03:
Hi,

First of all, thanks for releasing this SDK! I'm having lots of fun programming the genesis.

I'm trying to make a very simple platformer to learn genesis programming with SGDK. I'm using the high level API to create a map:
```
TILESET resTileMap1 "maps/map1.png" BEST DUPLICATE
MAP resMap1 "maps/map1.png" resTileMap1 BEST
```
I'm able to display the map and moving around. To detect where the player is I use `MAP_getTile` this way I know if the tile is floor a wall, etc.

But I'm unable to know how the tiles are ordered in memory. My current workflow is to compile, use the emulator to show the VRAM tiles and then hardcode the tile indexes in my code (floor tiles: 23, 26, 30, wall tiles: 34, 18, 33, etc). This doesn't feel right. It doesn't seem to load the tiles in order of appearance in the tileset png so I'm unable to see a better way.

What's the expected workflow?

closed 2021-04-04 by pladaria

Stephane-D, 2021-04-06:
Exactly they are stored in order of appearance in the image (top-left to right-bottom).
What you can do is to use 1 or 2 tile rows at the top just to order your tiles as you want then in your game logic you ignore those 1 or 2 tile rows by adding +8 or +16 offset on the Y position.

pladaria, 2021-04-06:
That was exactly my solution :)

Thanks again!