Stephane-D/SGDK #207: MAP using flipped tiles when tileset is defined using DUPLICATE

pladaria, 2021-04-12:
I've defined a tileset/map as:
```sh
TILESET resTileMap "maps/map.png" BEST DUPLICATE
MAP resMap "maps/map.png" resTileMap2 BEST
```
If I look at VRAM, everythings seems correct, flipped tiles have different tile entries:
[image]
For example, see the "ramp", going up and down. That's correct.

The problem is that the map seems to render using flipped tiles. When rendering a "ramp down", instead of using a "ramp down" tile, it uses a flipped "ramp up".

Stephane-D, 2021-04-12:
Thanks for reporting, i see why we have this issue ! I just pushed a fix to rescomp :)

closed 2021-04-12 by Stephane-D

Stephane-D, 2021-04-12:
Yeah, lucky timing :p