Stephane-D/SGDK #208: Kega-Fusion emulator gives the warning "Checksum incorrect" on rom.bin generated by SGDK

Oleg-N-Cher, 2021-04-24:
Hello dear SGDK developers,
I'm trying to develop on SGDK in Oberon language (using my Ofront+ Oberon-to-C translator).
And faced with such a problem that the the rom.bin created using the SGDK gives the warning "Checksum incorrect" in the Fusion emulator (https://www.carpeludum.com/kega-fusion/).

I'm in search of a third-party utility that can fix this problem by adjusting the checksum of the ROM image. Would you like to fix this on your side centrally? Thanks.

P.S. SGDK 1.62 (march 2021)

[image]

pladaria, 2021-04-24:
First result googling "fix sega genesis rom checksum";

https://www.romhacking.net/utilities/342/

Oleg-N-Cher, 2021-04-24:
Well, it works. Thank you.
To my regret, this utility is not for the command line, and it cannot be called from a .bat-file.
I also found a small utility SegaTool - https://www.romhacking.net/utilities/17/
which can fix checksum, but it is for MS-DOS.

andwn, 2021-04-25:
Unless you purposely include a routine in your game to validate the checksum, it doesn't matter what the checksum value is.

Here is a command line tool to set it anyway
https://github.com/sikthehedgehog/mdtools/tree/master/romfix

Oleg-N-Cher, 2021-04-25:
Thank you.
I specifically didn't include any code to check the checksum in my game. This is actually the example /sample/basics/hello-world/ from the distribution.
I still don't agree with you. This checksum is the sum of the ROM image. And this sum is checked by the emulator itself at the ROM loading stage. And from within the user code of a game, there is no way to validate the checksum, since it is part of the ROM image header and does not get into the user-level machine code.

andwn, 2021-04-25:
I'm saying the game will run on console without the checksum and Kega's warning is not relevant to homebrew. Console does not look at the checksum at boot. Most retail games have a validation built in as part of Sega's SDK library, but SGDK does not.

Oleg-N-Cher, 2021-04-25:
OK, I get you.

Then we reformulate the issue like this: is it required that the Kega Fusion emulator does not generate warnings about a corrupted checksum for ROMs generated by the SGDK?

If you think that it doesn't matter, and the SGDK is primarily focused on hardware, not emulators, then close this issue. Thank you.

andwn, 2021-04-25:
Kega is closed source and the most recent release was in 2010. This is normal for a period where ROM hacks were more popular than homebrew. Hacks would hang on a red screen with an invalid checksum.

(I don't develop SGDK. I just wanted to clarify how the checksums work.)

Oleg-N-Cher, 2021-04-26:
OK.
If the developers suddenly decide to fix this situation, then I see it as adding a utility to fix the ROM checksum and a new extension .bin into the makefile.gen to make corrected the checksum. And there, who wants to - can use it as an option.

Stephane-D, 2021-04-30:
Indeed the checksum warning is just a old thing from emulator because back in time some ROM dumps were corrupted, or incorrectly hacked. As @andwn said, SGDK does not include any checksum verification routine by default, it's up to the developer to do it if he want to have one. Eventually i made add a checksum fix to patch the header as it can _still_ be useful for people who want to do cart production for their homebrew.

Stephane-D, 2021-07-08:
I just rewrote _sizebnd_ tool which now add checksum to ROM. Note that the checksum is somehow custom to SGDK (optimized for speed) so it might be different from the one used in emulator. Again the checksum warning in emulator can be safely ignored.

closed 2021-07-08 by Stephane-D