Stephane-D/SGDK #209: MAP is rendering unexpected tiles

pladaria, 2021-04-30:
I tried to implement the most simple program reproducing this issue

### Project folder can be download here:
[issue1.zip](https://github.com/Stephane-D/SGDK/files/6405165/issue1.zip)
Player can be moved freely, issues use to appear when moving up and left

### main.c
```c
#include <genesis.h>

#include "../res/maps.h"
#include "../res/sprite.h"

Vect2D_s32 player = {
    .x = 50,
    .y = 600,
};

Vect2D_s32 camera = {
    .x = 0,
    .y = 0,
};

void handlePlayer() {
    u16 pad = JOY_readJoypad(JOY_1);
    if (pad & BUTTON_LEFT) player.x -= 8;
    if (pad & BUTTON_RIGHT) player.x += 8;
    if (pad & BUTTON_UP) player.y -= 8;
    if (pad & BUTTON_DOWN) player.y += 8;
    if (player.x < 0) player.x = 0;
    if (player.y < 0) player.x = 0;
}

void handleCamera() {
    camera.x = player.x - 150;
    camera.y = player.y - 150;
    if (camera.x < 0) camera.x = 0;
    if (camera.y < 0) camera.y = 0;
}

Sprite *playerSprite;
Map *map;

int main(bool hardReset) {
    // load map
    VDP_loadTileSet(&resTileMap, TILE_USERINDEX, CPU);
    map = MAP_create(&resMap, BG_A, TILE_ATTR_FULL(PAL2, FALSE, FALSE, FALSE, TILE_USERINDEX));
    // load sprites
    SPR_init();
    playerSprite = SPR_addSprite(&resSprite, player.x, player.y, TILE_ATTR(PAL3, TRUE, FALSE, FALSE));
    // init camera
    handleCamera();
    MAP_scrollTo(map, camera.x, camera.y);
    // apply palettes
    VDP_setPalette(PAL3, resSprite.palette->data);
    VDP_setPalette(PAL2, resMap.palette->data);
    SYS_showFrameLoad(TRUE);
    while (TRUE) {
        handlePlayer();
        handleCamera();
        MAP_scrollTo(map, camera.x, camera.y);
        SPR_setPosition(playerSprite, player.x - camera.x, player.y - camera.y);
        SPR_update();
        SYS_doVBlankProcess();
    }
    return 0;
}
```

### This is the map
[image]

### As you move around you will see strange tiles appearing:
[image]
[image]

### Video:
https://user-images.githubusercontent.com/579417/116693631-7cf14880-a9be-11eb-8fa5-080dc15561f1.mp4

Also, not sure if normal or related but I think it is strange to see garbage in plane while scrolling. I would expect to only see parts of the map:
[image]

Stephane-D, 2021-04-30:
Thanks for the complete example with videos !
Best test case ever :p
Definitely it looks like this is a bug happening in some situation..
The garbage in the plane explorer is expected, only visible part (with some small margin) of the scrolling is updated but for performance reason always a full line / column is transferred ending to some garbage in non visible areas. Not doing that I would have to double the number of DMA transfers, in fact my first implementation was doing that but it was a bit slower then.
I will test and try to fix asap !

Stephane-D, 2021-05-01:
Fixed ! Thanks again for the nice test case, really helped :)

closed 2021-05-01 by Stephane-D