Stephane-D/SGDK #210: Six-button vs 3-button detection is wrong

rfmerrill, 2021-05-01:
product spec: http://www.tmeeco.eu/SMD/spec_6Buttons_pad_rev2.pdf
technical bulletin: http://www.tmeeco.eu/SMD/Sega_Genesis_Manual/Bulletins/Gen-tech/Tech27-09.gif

The four bits that SGDK is using to distinguish between 3-button and 6-button are actually undefined, and I have what appears to be an official 6-button controller that does not consistently set these bits.

The bits that you are supposed to use are bits 0 and 1 of the third read with TH LOW. These will read 00 on a six-button, which would correspond to pressing up+down on a 3-button.

Stephane-D, 2021-05-03:
Thanks, after reviewing the changes, it seems that it should work (except in the specific UP+DOWN case but that would very rarely happen). What is weird is that from docs, 6 btn controllers should always return 1111 on fourth TH LOW state from the documents but your does not. Also i think it's the first time someone reported an issue with 6 button controller read code so i guess it's rare to have controller behaving as your ;)

rfmerrill, 2021-05-03:
I have two SEGA 6-button controllers and one third-party one, and the different behavior only occurs on one of the SEGA 6 button controllers. If you hold X+Z+B+C then it changes from 1111 to 1100. I have no idea how many controllers behave like this, it's likely it's a small minority and it's a very specific situation where it comes up.

I discovered this because I noticed that mashing buttons on the controller caused 240p test suite's controller test to switch to 3-button. I first approached the problem as the controller being defective or counterfeit, but when I told Tiido about it they pointed out that the check in the code was wrong.

Stephane-D, 2021-05-03:
I merged your changes but i actually even slightly modified it to take care of that. Good to know it still show 1100 as this is exactly what i used to recognize it (make some sense after all) 👍

closed 2021-05-03 by Stephane-D

rfmerrill, 2021-05-03:
I think your additional commit might also be problematic... the technical bulletin says that the 1111 bits should not be used at all (I think? It says "shaded" and the scan makes it hard to tell). There may be other controllers out there that set those bits to other random combinations.

If you really wanted to handle the up+down case perfectly... I don't know if there's a way to do it that follows the letter of the spec. The closest thing I can think of is to check that the up/down bits are 00 in v2, and _not_ 00 in the other reads. But even then, they could be holding up+down on a six-button controller.

I think the up+down case is really quite minor, as all that's really needed is to release that combination on the controller, the pad type will change to 3-button and then it won't change back to 6 unless `JOY_reset` is called while up+down are again held. It's likely this scenario would also cause misbehavior on real genesis games.

Stephane-D, 2021-05-03:
I do not see how my additional changes can be problematic as i'm just testing them if first condition wasn't enough to ensure we have a 3 button controller : i'm just testing bit for 0 value (not for 1 afaik) to confirm we have a 3 button controller whereas these bits have great luck to be 1 on a 6 button controller (and that was how we tested 6 button controller initially) so normally it just should help for the UP+DOWN case

rfmerrill, 2021-05-03:
Well, no use being worried about a problem that hasn't happened yet, but it's possible there are 6-button controllers out there that set those bits to zero. Given how I'm the first person to encounter this issue in general it could very well be that these are never set to 00 on a 6-button. In case it comes up, though, the conversation in this ticket could be good to help solve it. Thank you for responding to my PR and merging it :)

Stephane-D, 2021-05-03:
Indeed, but i think that at least that would help to solve the up+down case, i think having both cases at same time (up+down and 6 buttons returning 00xx instead of 11xx) is very unlikely to happen anyway.

Stephane-D, 2021-05-03:
Thanks to you for your contribution ! Honestly that issue would have be difficult to catch for me as i didn't know 6 buttons controllers could behave like this !

rfmerrill, 2021-05-03:
> Indeed, but i think that at least that would help to solve the up+down case, i think having both cases at same time (up+down and 6 buttons returning 00xx instead of 11xx) is very unlikely to happen anyway.

I'm sorry if I'm not understanding correctly, but your commit has it setting the controller to 3-button if it detects _either_ condition, so if a 6-button controller returns 00xx it will always be detected as 3-button, right?

Stephane-D, 2021-05-03:
Oh yeah indeed but i really believe that 6 btn controller returning 00xx is really unlikely to happen anyway, again you were the first to report the issue, and yourself you observed that your actually return 11xx at least, and in some way, that makes sense.
I'm afraid about the UP+DOWN issue to be a real problem on emulators used with keyboard for instance (in which case XYZ won't be taken in account)