Stephane-D/SGDK #212: Roms generated by SGDK 1.62 bigger than 2MB won't load on Everdrive X3

heitorgcosta, 2021-05-12:
Any Rom that I try to build that ends up bigger than 2MB with SGDK 1.62 will work fine on emulators but won't boot up on an Everdrive X3, it'll get stuck on "loading". I'm not sure about other versions of Everdrive, as I don't know anyone else that has an Everdrive.
Building the same projects with SGDK 1.60 makes the Roms bootable on the Everdrive X3 again.

[Here](https://github.com/heitorgcosta/sgdk-everdrive-x3-issue) is a sample project that if built with 1.62 will generate an unbootable rom (on the everdrive) but works fine with 1.60.

Having people to also test this on different versions of everdrive would be nice.

dleslie, 2021-05-12:
Do **_any_** ROMs bigger than 2MB load on the X3?

pladaria, 2021-05-13:
I compiled your sample project and it works on my cheap aliexpress flashcart
This one: https://es.aliexpress.com/item/1005002294163237.html

heitorgcosta, 2021-05-13:
It might be a thing specifically with Everdrives then, but I'm not sure if it's specific to the X3. I have an X7 on the way, but it might take a while to arrive and I'm also trying to find people that own other Everdrives to test as well.

I'm assuming it might have to do with the whole bank switching changes added to 1.62 and how the Everdrive handles bank switching, but I'm not sure, since it can properly run SSF2 which uses bank switching.

heitorgcosta, 2021-05-13:
Yeah, it turns out that it's being caused by the fix introduced at the 85200b3d46234676586e5e723232943405d7ad13 commit.

```
// reset banks as we messed them during init
len = 8;
while(--len) SYS_setBank(len, len);
```

Rebulding with this code commented makes the rom bootable again on the Everdrive X3.

Stephane-D, 2021-05-17:
Thanks for notifying about the problem. I will investigate it to see why it gives issues !

Stephane-D, 2021-05-17:
Looking at the code i don't see why it fails only with ROM are > 2MB.
I think there is an issue with the way the X3 is handling SRAM (which stay in the > 2MB region).
For the sake of safety, i put that code only enabled if bank switch is enabled (which make sense). I hope that is enough to fix the issue with X3, it would be nice to test if bank switch does work though when it's enabled for ROM > 4MB.

closed 2021-05-17 by Stephane-D