Stephane-D/SGDK #213: Wrong tile generation by rescomp

pladaria, 2021-05-12:
Hi, 

I'm having issues with some PNGs, it looks like the tile extraction has some problem

:point_right: Download project files: [issue2.zip](https://github.com/Stephane-D/SGDK/files/6468309/issue2.zip)

### main.c
```c
#include <genesis.h>

#include "../res/bg.h"

int main(bool hardReset) {
    VDP_setScreenWidth320();
    VDP_setScreenHeight224();
    VDP_setPlaneSize(64, 32, TRUE);

    VDP_drawImageEx(BG_B, &resBgSea, TILE_ATTR_FULL(PAL1, 0, 0, 0, TILE_USERINDEX), 0, 0, FALSE, FALSE);
    VDP_setPalette(PAL1, resBgSea.palette->data);

    while (TRUE) {
        SYS_doVBlankProcess();
    }

    return 0;
}
```
### Source image
[image]
 
### Result
[image]

### VDP:
[image]

heitorgcosta, 2021-05-12:
I'd assume the issue is the bit depth, which GIMP seems to have exported as 2 bits. Opening the picture through Aseprite and exporting it as an indexed png (8-bit depth) made it so that the image loads correctly with SGDK.

pladaria, 2021-05-13:
Yep, forcing the image to 8bpp solves the issue, so there must be something wrong in the PNG bitstream reading when the samples are stored at 2bpp

matteusbeus, 2021-05-13:
yes I've had this problem previously!

alicesim1, 2021-05-13:
It turns out that the PNG file; if it is 4 colors, but it is exported by GIMP and it is not fully compatible, only partially with Rescom.

If the PNG is opened for example with Aseprite and the PNG is saved is 4 colors, again if it works perfectly with Rescom.

heitorgcosta, 2021-05-13:
Did a bit of digging and it turns out the `ImageUtil.java` file has a typo specifically on the function that converts 2bpp to 8bpp that rescomp calls.

```java
static byte[] convert2bppTo8bpp(byte[] data)
{
    final byte[] result = new byte[data.length * 4];

    for (int i = 0; i < data.length; i++)
    {
        result[(i * 4) + 0] = (byte) ((data[i] >> 6) & 0x03);
        result[(i * 4) + 1] = (byte) ((data[i] >> 4) & 0x03);
        result[(i * 4) + 2] = (byte) ((data[i] >> 3) & 0x03); <- Should shift by 2 instead of 3
        result[(i * 4) + 3] = (byte) ((data[i] >> 0) & 0x03);
    }

    return result;
}
```

Adjusting it and re-exporting the rescomp jar fixed the issue.

pladaria, 2021-05-13:
> Did a bit of digging and it turns out the `ImageUtil.java` file has a typo specifically on the function that converts 2bpp to 8bpp that rescomp calls.
>
> ```java
> static byte[] convert2bppTo8bpp(byte[] [...]

good catch!

Stephane-D, 2021-05-17:
Indeed good catch ! Just fixed :) Thanks for pointing it 👍

closed 2021-05-17 by Stephane-D