Stephane-D/SGDK #217: Page fault trying to build samples/sonic under Docker

franciscod, 2021-06-11:
```$ docker run --rm -v "$PWD":/src sgdk
mkdir -p src/boot
mkdir -p out
cp /sgdk/src/boot/sega.s src/boot/sega.s
cp /sgdk/src/boot/rom_head.c src/boot/rom_head.c
/sgdk/bin/gcc -m68000 -Wall -Wextra -Wno-shift-negative-value -Wno-main -Wno-unused-parameter -fno-builtin -Iinc -Isrc -Ires -I/sgdk/inc -I/sgdk/res -B/sgdk/bin -c src/boot/rom_head.c -o out/rom_head.o
wine: created the configuration directory '/sgdk/bin/.wineconf'
Could not load wine-gecko. HTML rendering will be disabled.
wine: configuration in '/sgdk/bin/.wineconf' has been updated.
/sgdk/bin/ld -T /sgdk/md.ld -nostdlib --oformat binary -o out/rom_head.bin out/rom_head.o
/sgdk/bin/gcc -x assembler-with-cpp -m68000 -Wall -Wextra -Wno-shift-negative-value -Wno-main -Wno-unused-parameter -fno-builtin -Iinc -Isrc -Ires -I/sgdk/inc -I/sgdk/res -B/sgdk/bin -c src/boot/sega.s -o out/sega.o
java -jar /sgdk/bin/rescomp.jar res/gfx.res res/gfx.rs -dep out/res/gfx.o
ResComp 3.34 - SGDK Resource Compiler - Copyright 2021 (Stephane Dallongeville)

Resource: PALETTE palette_all "gfx/S1_GHZ1_FG.png"
--> executing plugin PALETTE...'palette_all' raw size: 134 bytes
Resource: TILESET bga_tileset "gfx/S1_GHZ1_FG.png" BEST ALL
--> executing plugin TILESET...'bga_tileset' raw size: 14440 bytes
Resource: TILESET bgb_tileset "gfx/S1_GHZ1_BG.png" BEST ALL
--> executing plugin TILESET...'bgb_tileset' raw size: 6984 bytes
Resource: MAP bga_map "gfx/S1_GHZ1_FG.png" bga_tileset BEST 0
--> executing plugin MAP...MAP 'bga_map' details: 353 metatiles, 132 blocks, block grid size = 80 x 10 - optimized = 80 x 10
'bga_map' raw size: 20706 bytes
Resource: MAP bgb_map "gfx/S1_GHZ1_BG.png" bgb_tileset BEST 0
--> executing plugin MAP...MAP 'bgb_map' details: 148 metatiles, 28 blocks, block grid size = 64 x 2 - optimized = 64 x 2
'bgb_map' raw size: 3274 bytes

'bga_tileset_data' packed with APLIB, size = 5459 (38% - origin size = 14432)
'bgb_tileset_data' packed with APLIB, size = 2665 (38% - origin size = 6976)
'bga_map_metatiles' packed with APLIB, size = 2053 (73% - origin size = 2824)
'bga_map_mapBlocks' packed with APLIB, size = 3934 (23% - origin size = 16896)
'bga_map_mapBlockIndexes' packed with APLIB, size = 294 (37% - origin size = 800)
'bgb_map_metatiles' packed with APLIB, size = 915 (77% - origin size = 1184)
'bgb_map_mapBlocks' packed with APLIB, size = 987 (55% - origin size = 1792)
'bgb_map_mapBlockIndexes' packed with APLIB, size = 65 (51% - origin size = 128)

res/gfx.res summary:
-------------
Binary data: 16658 bytes
  Unpacked: 280 bytes
  Packed: 16378 bytes (36% - origin size: 45032 bytes)
Misc metadata (bitmap, image, tilemap, tileset, palette..): 34 bytes
Total: 16692 bytes (16 KB)
mkdir -p out/res/
/sgdk/bin/gcc -x assembler-with-cpp -m68000 -Wall -Wextra -Wno-shift-negative-value -Wno-main -Wno-unused-parameter -fno-builtin -Iinc -Isrc -Ires -I/sgdk/inc -I/sgdk/res -B/sgdk/bin -O3 -fuse-linker-plugin -fno-web -fno-gcse -fno-unit-at-a-time -fomit-frame-pointer -flto -c res/gfx.rs -o out/res/gfx.o
cp res/gfx.d out/res/gfx.d
rm res/gfx.d
java -jar /sgdk/bin/rescomp.jar res/sound.res res/sound.rs -dep out/res/sound.o
ResComp 3.34 - SGDK Resource Compiler - Copyright 2021 (Stephane Dallongeville)

Resource: XGM sonic_music "music/sonic1.vgm"
--> executing plugin XGM...Executing /sgdk/bin/xgmtool /src/res/music/sonic1.vgm /src/res/music/sonic1.bin -s   
Warning: option  not recognized (ignored)
Warning: option  not recognized (ignored)
'sonic_music' raw size: 21760 bytes
Resource: WAV sonic_jump_sfx "sfx/S1_jump.wav" XGM
--> executing plugin WAV...'sonic_jump_sfx' raw size: 6144 bytes
Resource: WAV sonic_roll_sfx "sfx/S1_roll.wav" XGM
--> executing plugin WAV...'sonic_roll_sfx' raw size: 28416 bytes
Resource: WAV sonic_stop_sfx "sfx/S1_stop.wav" XGM
--> executing plugin WAV...'sonic_stop_sfx' raw size: 6656 bytes

res/sound.res summary:
-------------
Binary data: 62976 bytes
  Unpacked: 62976 bytes
Total: 62976 bytes (62 KB)
mkdir -p out/res/
/sgdk/bin/gcc -x assembler-with-cpp -m68000 -Wall -Wextra -Wno-shift-negative-value -Wno-main -Wno-unused-parameter -fno-builtin -Iinc -Isrc -Ires -I/sgdk/inc -I/sgdk/res -B/sgdk/bin -O3 -fuse-linker-plugin -fno-web -fno-gcse -fno-unit-at-a-time -fomit-frame-pointer -flto -c res/sound.rs -o out/res/sound.o
cp res/sound.d out/res/sound.d
[... 172 lines cut ...]
'gbar_sprite_animation0_frame14_tileset_data' packed with LZ4W, size = 32 (33% - origin size = 96)
'gbar_sprite_animation0_frame15_tileset_data' packed with LZ4W, size = 28 (29% - origin size = 96)
'gbar_sprite_animation0_frame16_tileset_data' packed with LZ4W, size = 28 (29% - origin size = 96)

res/sprite.res summary:
-------------
Binary data: 12268 bytes
  Unpacked: 96 bytes
  Packed: 12172 bytes (59% - origin size: 20736 bytes)
Sprite metadata (all but tiles and palette data): 1974 bytes
Misc metadata (bitmap, image, tilemap, tileset, palette..): 650 bytes
Total: 14892 bytes (15 KB)
mkdir -p out/res/
/sgdk/bin/gcc -x assembler-with-cpp -m68000 -Wall -Wextra -Wno-shift-negative-value -Wno-main -Wno-unused-parameter -fno-builtin -Iinc -Isrc -Ires -I/sgdk/inc -I/sgdk/res -B/sgdk/bin -O3 -fuse-linker-plugin -fno-web -fno-gcse -fno-unit-at-a-time -fomit-frame-pointer -flto -c res/sprite.rs -o out/res/sprite.o
cp res/sprite.d out/res/sprite.d
rm res/sprite.d
mkdir -p out/src/
/sgdk/bin/gcc -m68000 -Wall -Wextra -Wno-shift-negative-value -Wno-main -Wno-unused-parameter -fno-builtin -Iinc -Isrc -Ires -I/sgdk/inc -I/sgdk/res -B/sgdk/bin -O3 -fuse-linker-plugin -fno-web -fno-gcse -fno-unit-at-a-time -fomit-frame-pointer -flto -MMD -c src/main.c -o out/src/main.o
echo "out/res/gfx.o out/res/sound.o out/res/sprite.o out/src/main.o" > out/cmd_
/sgdk/bin/gcc -B/sgdk/bin -n -T /sgdk/md.ld -nostdlib out/sega.o @out/cmd_ /sgdk/lib/libmd.a /sgdk/lib/libgcc.a -o out/rom.out -Wl,--gc-sections
wine: Unhandled page fault on read access to 0x00000000 at address 0x490ebd (thread 004c), starting debugger...
Can't attach process 004b: error 5
make: *** [/sgdk/makefile.gen:171: out/rom.out] Error 1
rm res/gfx.rs res/sprite.rs res/sound.rs
```

pladaria, 2021-06-11:
Can you provide some more context?

- What is your host operating system and architecture?
- What docker version are you using?
- Any other relevant information

franciscod, 2021-06-11:
- Debian 10 (buster), amd64
- Docker version 18.09.1, build 4c52b90
- Yep, only that some part of the output isn't there (I guess there's not much to build now). Here's the output of the same command:
```
$ docker run --rm -v "$PWD":/src sgdk
mkdir -p src/boot
mkdir -p out
/sgdk/bin/gcc -B/sgdk/bin -n -T /sgdk/md.ld -nostdlib out/sega.o @out/cmd_ /sgdk/lib/libmd.a /sgdk/lib/libgcc.a -o out/rom.out -Wl,--gc-sections
wine: created the configuration directory '/sgdk/bin/.wineconf'
Could not load wine-gecko. HTML rendering will be disabled.
wine: configuration in '/sgdk/bin/.wineconf' has been updated.
wine: Unhandled page fault on read access to 0x00000000 at address 0x490ebd (thread 0024), starting debugger...
Can't attach process 0023: error 5
make: *** [/sgdk/makefile.gen:171: out/rom.out] Error 1
```

franciscod, 2021-06-11:
Please let me know if I can try anything else here! Glad to help.

pladaria, 2021-06-12:
That's strange, just to discard things, could you try with a newer docker version?
I'm using 
```
$ docker --version
Docker version 20.10.2, build 20.10.2-0ubuntu1~20.04.2
```

pladaria, 2021-06-12:
Would you mind to try this draft PR?

It uses a 64bit base image:
https://github.com/Stephane-D/SGDK/pull/218

franciscod, 2021-06-15:
Looking good! Same docker version but this thime with the 64bit fork:

```
$ docker run --rm -v "$PWD":/src sgdk-64
mkdir -p src/boot
mkdir -p out
/sgdk/bin/gcc -B/sgdk/bin -n -T /sgdk/md.ld -nostdlib out/sega.o @out/cmd_ /sgdk/lib/libmd.a /sgdk/lib/libgcc.a -o out/rom.out -Wl,--gc-sections
wine: created the configuration directory '/tmp/wine/.wineconf'
Could not find Wine Gecko. HTML rendering will be disabled.
wine: configuration in L"/tmp/wine/.wineconf" has been updated.
rm out/cmd_
/sgdk/bin/objcopy -O binary out/rom.out out/rom.bin
/sgdk/bin/sizebnd out/rom.bin -sizealign 131072
/sgdk/bin/nm --plugin=liblto_plugin-0.dll -n out/rom.out > out/symbol.txt
```

closed 2021-07-08 by Stephane-D