Stephane-D/SGDK #220: Docker (v3.5.1.7) OSX (11.3) not working for me

Quentie, 2021-07-07:
I tried this evening to get the sonic example to compile for me using docker (v3.5.1.7) on OSX (11.3).
Unfortunately I had no success :( Does anyone maybe have a hint what I might be doing wrong? 

```
docker run --rm -v "$PWD":/src sgdk
WARNING: The requested image's platform (linux/386) does not match the detected host platform (linux/amd64) and no specific platform was requested
mkdir -p src/boot
mkdir -p out
/sgdk/bin/gcc -m68000 -Wall -Wextra -Wno-shift-negative-value -Wno-main -Wno-unused-parameter -fno-builtin -Iinc -Isrc -Ires -I/sgdk/inc -I/sgdk/res -B/sgdk/bin -c src/boot/rom_head.c -o out/rom_head.o
wine: created the configuration directory '/sgdk/bin/.wineconf'
make: *** [/sgdk/makefile.gen:183: out/rom_head.o] Error 1
```

pladaria, 2021-07-07:
Would you mind to try with the docker image from this PR https://github.com/Stephane-D/SGDK/pull/218 ?

Quentie, 2021-07-07:
Thanks a lot @pladaria for your quick reply!
This works like a charm for me!

Stephane-D, 2021-07-08:
MR has been merged

closed 2021-07-08 by Stephane-D