Stephane-D/SGDK #221: PAL_fadeToAll() causes CRAM dots

a-dietrich, 2021-08-02:
I've been trying to fade all 64 colors at once using PAL_fadeToAll(). Unfortunately, this causes some CRAM dots to appear in the middle of the screen for one frame. 

Attached is some sample code as well as a test rom. The issue can be tested on are real machine or with BlastEm:

[image]

[cramdots.zip](https://github.com/Stephane-D/SGDK/files/6919701/cramdots.zip)

Looking through the SGDK code, it appears that the problem might be in setFadePalette():

```
static void setFadePalette(u16 ind, const u16 *src, u16 len)
{
    static u32 lastVTimer = 0;

    // be sure to wait at least 1 frame between set fade palette call
    if (lastVTimer == vtimer) SYS_doVBlankProcess();

    // use DMA for long transfer
    PAL_setColors(ind, src, len, (len > 16)?DMA:CPU);

    // keep track of last update
    lastVTimer = vtimer;
}
```

When this gets called in the first fading step, lastVTimer and vtimer may differ. In that case SYS_doVBlankProcess() would not not get called.

Stephane-D, 2021-08-03:
Thanks ! Indeed on the first call it may not wait for VBlank, i think i simplified the code recently and broke that. I just pushed the fix :)
Also be careful with the PAL_fadeToxxx(..) methods as they do **read CRAM** (to get current palette) before processing the fade operation and read CRAM does produce CRAM dots too, so it's important to call the PAL_fadeToxx(..) method during VBlank (SGDK won't do it by itself as it may eventually be undesired for async process).

closed 2021-08-03 by Stephane-D

a-dietrich, 2021-08-07:
Awesome. Just tried it. Thanks for fixing!