Stephane-D/SGDK #227: MD MSU MD+ audio support

PlaymanBR, 2021-11-01:
Hi, Is it possible and how can i implement the MD MSU MD+ audio in games?

Stephane-D, 2021-11-02:
Hi,

SGDK does not provide any _assistance_ with the MD MSU / MD+ features of some flash carts nor MCD in general so you should handle that by yourself. You can find some documentation here about how use the MSU MD:
https://github.com/krikzz/msu-md

I think it can be used with SGDK, at least you can use SGDK compiler to compile anything from the MSU-MD sdk :)

closed 2021-11-02 by Stephane-D

PlaymanBR, 2021-11-07:
Hi, thanks for the reply.

We are players who dreamed of having arcade versions like SNK / KOF 9x ,
Marvel fighting and Capcom / Street Fighter Alpha games series for genesis.

Please, could you know the work of my brazilian fighting game development
team for genesis?

I am responsible for the development of the audio area.

We are also developing tools that speed up sprite production and game
physics.

Here's our youtube channel for you to check it out.

https://www.youtube.com/playlist?list=PLO-5vY5uaWRt8VUzMrCrAj_BeCtK3PDya

We would be very happy if you could leave us comments and mention our
projects in your portfolio.

*Thank you very much!!  YOU and the SGDK has opened up new avenues for sega
genesis lovers. :))*

Stephane-D, 2021-11-15:
Your projects looks quite promising, it would be really nice to see a complete game out of it though :)
In the current state it's more of a POC than a real game (it misses important parts as collisions) and to be honest lot of that exists so i cannot reference all them.

Lately you had this POC also showing vs fighters animations (ok, that is far less advanced but animation is impressive by itself) :
https://twitter.com/masterlinkueibr/status/1460054520969449476
https://twitter.com/masterlinkueibr/status/1459884704702816260

PlaymanBR, 2021-11-20:
Thanks for the feedback!