Stephane-D/SGDK #231: Suddenly SGDK refuses to compile and run the bin, and for some reason resources.h has a squiggly line underneath it.

thinkOfAnAnswer, 2021-12-10:
Hey folks,

Sorry for the lack of organization, new to both SGDK and GitHub. Anyways, I was doing some tutorials and everything worked fine, but out of nowhere I got a squiggly line under resources, have not been able to compile and run the emulator from the ctrl+shift+p shortcut, and I get this massive list of issues in the Terminal. What am I not understanding here that is causing all this, because I haven't really deleted or tinkered with anything.

Thank you for your time, cheers.

[inside Terminal text below]

C:/SGDK/bin/gcc.exe -m68000 -Wall -Wextra -Wno-shift-negative-value -Wno-main -Wno-unused-parameter -fno-builtin -Iinc -Isrc -Ires -IC:/SGDK/inc -IC:/SGDK/res -BC:/SGDK/bin -O3 -fuse-linker-plugin -fno-web -fno-gcse -fno-unit-at-a-time -fomit-frame-pointer -flto -MMD -c src/main.c -o out/src/main.o
src/main.c:9:0: warning: ignoring #pragma region  [-Wunknown-pragmas]
 #pragma region

src/main.c:45:0: warning: ignoring #pragma endregion  [-Wunknown-pragmas]
 #pragma endregion

src/main.c:48:0: warning: ignoring #pragma region  [-Wunknown-pragmas]
 #pragma region

src/main.c: In function 'endGame':
src/main.c:85:18: warning: passing argument 1 of 'showText' discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
         showText(msgReset);
                  ^~~~~~~~
src/main.c:49:6: note: expected 'char *' but argument is of type 'const char *'      
 void showText(char s[]) // display a line of text at center of screen
      ^~~~~~~~
src/main.c: In function 'joystickHandler':
src/main.c:90:6: warning: type of 'joy' defaults to 'int' [-Wimplicit-int]
 void joystickHandler(joy, changed, state)
      ^~~~~~~~~~~~~~~
src/main.c:90:6: warning: type of 'changed' defaults to 'int' [-Wimplicit-int]       
src/main.c:90:6: warning: type of 'state' defaults to 'int' [-Wimplicit-int]
src/main.c: At top level:
src/main.c:115:0: warning: ignoring #pragma endregion  [-Wunknown-pragmas]
 #pragma endregion

src/main.c: In function 'main':
src/main.c:121:25: warning: passing argument 1 of 'JOY_setEventHandler' from incompatible pointer type [-Wincompatible-pointer-types]
     JOY_setEventHandler(&joystickHandler);
                         ^
In file included from C:/SGDK/inc/genesis.h:47:0,
                 from src/main.c:1:
C:/SGDK/inc/joy.h:182:6: note: expected 'void (*)(u16,  u16,  u16) {aka void (*)(short unsigned int,  short unsigned int,  short unsigned int)}' but argument is of type 'void (*)()'
 void JOY_setEventHandler(JoyEventCallback *CB);
      ^~~~~~~~~~~~~~~~~~~
src/main.c:130:0: warning: ignoring #pragma region  [-Wunknown-pragmas]
     #pragma region

src/main.c:156:0: warning: ignoring #pragma endregion  [-Wunknown-pragmas]
     #pragma endregion

src/main.c:159:0: warning: ignoring #pragma region  [-Wunknown-pragmas]
     #pragma region

src/main.c:168:0: warning: ignoring #pragma endregion  [-Wunknown-pragmas]
     #pragma endregion

src/main.c:170:14: warning: passing argument 1 of 'showText' discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
     showText(msgStart);
              ^~~~~~~~
src/main.c:49:6: note: expected 'char *' but argument is of type 'const char *'      
 void showText(char s[]) // display a line of text at center of screen
      ^~~~~~~~
src/main.c:176:0: warning: ignoring #pragma region  [-Wunknown-pragmas]
         #pragma region

src/main.c:218:0: warning: ignoring #pragma endregion  [-Wunknown-pragmas]
         #pragma endregion

echo "out/res/resources.o out/src/main.o" > out/cmd_
C:/SGDK/bin/gcc.exe -BC:/SGDK/bin -n -T C:/SGDK/md.ld -nostdlib out/sega.o @out/cmd_ 
C:/SGDK/lib/libmd.a C:/SGDK/lib/libgcc.a -o out/rom.out -Wl,--gc-sections
C:/SGDK/bin/rm.exe out/cmd_
C:/SGDK/bin/objcopy.exe -O binary out/rom.out out/rom.bin
java -jar C:/SGDK/bin/sizebnd.jar out/rom.bin -sizealign 131072 -checksum
92642
java.lang.ArrayIndexOutOfBoundsException: 92642
        at sgdk.sizebnd.Launcher.getInt(Launcher.java:134)
        at sgdk.sizebnd.Launcher.getChecksum(Launcher.java:145)
        at sgdk.sizebnd.Launcher.setChecksum(Launcher.java:159)
        at sgdk.sizebnd.Launcher.execute(Launcher.java:79)
        at sgdk.sizebnd.Launcher.main(Launcher.java:59)
make: *** [out/rom.bin] Error 1
rm res/resources.rs

thinkOfAnAnswer, 2021-12-10:
Also, Stephen. I'm an indie dev and would love to be able to make a future game on the Mega Drive alongside modern systems for my future projects, and using the MEGA CD in some capacity would be a dream, so is there any chance of an expansion to the SGDK library to be compatible with the MEGA CD in the future?

With Terraonion's Mega SD and Krikzz EverDrive, development and testing on actual hardware is now a genuine reality, so I'm curious to know what you're thinking there.

Finally, what is realistically, the largest size a Mega Drive rom file can be? With EverDrives, we're no longer confined to 4MBs, so what would the scientific limit be? I'm asking in case the MEGA CD option is out, we could just take the SNES EverDrive SD card extra capacity approach if that's possible.

Sorry for all the questions, but its been a very exciting week for me working with this thing and a life dream is now a genuine reality thanks to you!

Stephane-D, 2021-12-13:
Hi @thinkOfAnAnswer 

I don't know which tutorial you followed but the _#pragma region_ error you got is really weird for the least, does your code really contains _#pragma_ keyword ? this is really low level C stuff and you shouldn't need that with SGDK.

For your second question, lot of people already are asking for MCD support in SGDK. I will be honest, this is not a priority feature for me. I eventually will add limited support to it (just to use CDDA for instance) but i still give priority on Sega Mega Drive / Genesis first. 4 MB is the maximum without involving bank switching mechanism, with bank switching (SGDK provides SSF2 mapper support) you can go higher than that, i would say the theoretical limit is really high (128 MB) but realistically when you're using the SGDK embedded mapper support, 16 MB is probably the maximum you can get (because all common data and all resources metadata have still to stick in 3 MB limit).

If you want support about SGDK in general, you 'd better join the [SGDK Discord server](https://discord.gg/xmnBWQS)
Or use the [SpritesMind forum](http://gendev.spritesmind.net/forum/) if you don't use Discord.

As _Github Issues_ are more about submitting bug with SGDK :)

closed 2021-12-13 by Stephane-D

thinkOfAnAnswer, 2021-12-13:
Hey Stephen,

I'm using the OhSat tutorial list (MegaRunner project), but the game runs and compiles fine, its this specific project that won't compile and run from within VSC. 

Can the pragma region lead to issues like that? Why is ignoring it? Does that also include the code inside it?

"this is really low level C stuff and you shouldn't need that with SGDK."

I'm fairly new to C so I thought I'd organize and make things cleaner for me to see and read. What if my  project increases in scope and things get a bit on the messy end (I get lost when the page is full of code and you have to keep scrolling down), how would I make things in smaller chunks? Would I write the entire project in the main.c file? I guess what I'm asking is, how would you divide this into smaller chunks?

- In that case, is it theoretically possible to let the MCD handle all the audio tracks to save on cartridge space when that feature is made? The addition of cutscenes is out of the question I take it?

I'll be sure to join the Discord community, sounds like a great resource to lean into.

Thank you so much for your time, Stephen!