Stephane-D/SGDK #240: Megapack Compress in Rescomp

saintienn, 2022-01-10:
I found this graphic compressor for mega drive (https://github.com/lab313ru/megapack-megadrive), but I have no idea how to integrate it on Rescomp. I believe that is very interesting how an alternative for heavy compress of  tiles and maps.

Stephane-D, 2022-01-10:
There is an example of using it with SGDK here :
https://github.com/lab313ru/megapack-megadrive/tree/main/sega_test
So basically just by taking the example you should be able to use it, still they do something weird as it seems to use a block of memory without properly allocating it. I will try to look at it if the compressor is freely usable / available but i'm not sure..

saintienn, 2022-01-11:
Thank you so much!!

closed 2022-01-11 by Stephane-D