Stephane-D/SGDK #241: libres.h missing - compiling sgdk fails?

dantel35, 2022-02-05:
Hi,
it seems that compiling the newest version of SGDK fails.

Here is what the output says:
`C:\sgdk>C:\sgdk\bin\make -f C:\sgdk\makefile.gen
C:/sgdk/bin/mkdir.exe -p src/boot
C:/sgdk/bin/mkdir.exe -p out
C:/sgdk/bin/gcc.exe -m68000 -Wall -Wextra -Wno-shift-negative-value -Wno-main -Wno-unused-parameter -fno-builtin -Iinc -Isrc -Ires -IC:/sgdk/inc -IC:/sgdk/res -BC:/sgdk/bin -c src/boot/rom_head.c -o out/rom_head.o
C:/sgdk/bin/ld.exe -T C:/sgdk/md.ld -nostdlib --oformat binary -o out/rom_head.bin out/rom_head.o
C:/sgdk/bin/gcc.exe -x assembler-with-cpp -m68000 -Wall -Wextra -Wno-shift-negative-value -Wno-main -Wno-unused-parameter -fno-builtin -Iinc -Isrc -Ires -IC:/sgdk/inc -IC:/sgdk/res -BC:/sgdk/bin -c src/boot/sega.s -o out/sega.o
java -jar C:/sgdk/bin/rescomp.jar res/libres.res res/libres.rs -dep out/res/libres.o
ResComp 3.36 - SGDK Resource Compiler - Copyright 2021 (Stephane Dallongeville)

Resource: NEAR
--> executing plugin NEAR...
Resource: BITMAP sgdk_logo "image/sgdk_logo.png" -1
--> executing plugin BITMAP...'sgdk_logo' raw size: 2100 bytes
Resource: TILESET font_default "image/font_default.png" BEST NONE
--> executing plugin TILESET...'font_default' raw size: 3080 bytes
Resource: PALETTE font_pal_default "image/font_default.png"
--> executing plugin PALETTE...'font_pal_default' raw size: 38 bytes
Resource: BIN stop_xgm "sound/stop_xgm.bin" 2 2 0 NONE FALSE
--> executing plugin BIN...'stop_xgm' raw size: 172 bytes

'font_default_data' packed with APLIB, size = 609 (20% - origin size = 3072)
'sgdk_logo_data' packed with APLIB, size = 1081 (53% - origin size = 2048)

res/libres.res summary:
-------------
Binary data: 1928 bytes
  Unpacked: 236 bytes
  Packed: 1692 bytes (33% - origin size: 5120 bytes)
Misc metadata (bitmap, image, tilemap, tileset, palette..): 34 bytes
Total: 1962 bytes (2 KB)
C:/sgdk/bin/mkdir.exe -p out/res/
C:/sgdk/bin/gcc.exe -x assembler-with-cpp -m68000 -Wall -Wextra -Wno-shift-negative-value -Wno-main -Wno-unused-parameter -fno-builtin -Iinc -Isrc -Ires -IC:/sgdk/inc -IC:/sgdk/res -BC:/sgdk/bin -O3 -fuse-linker-plugin -fno-web -fno-gcse -fno-unit-at-a-time -fomit-frame-pointer -flto -c res/libres.rs -o out/res/libres.o
C:/sgdk/bin/cp.exe res/libres.d out/res/libres.d
C:/sgdk/bin/rm.exe res/libres.d
C:/sgdk/bin/sjasm.exe -isrc -iinc -ires -iC:/sgdk/src -iC:/sgdk/inc src/z80_drv0.s80 src/z80_drv0.o80 out.lst
SjASM Z80 Assembler v0.39g6 - www.xl2s.tk
Pass 1 complete (0 errors)
Pass 2 complete
Errors: 0
C:/sgdk/bin/bintos.exe src/z80_drv0.o80
C:/sgdk/bin/mkdir.exe -p out/src/
C:/sgdk/bin/gcc.exe -x assembler-with-cpp -m68000 -Wall -Wextra -Wno-shift-negative-value -Wno-main -Wno-unused-parameter -fno-builtin -Iinc -Isrc -Ires -IC:/sgdk/inc -IC:/sgdk/res -BC:/sgdk/bin -O3 -fuse-linker-plugin -fno-web -fno-gcse -fno-unit-at-a-time -fomit-frame-pointer -flto -MMD -c src/z80_drv0.s -o out/src/z80_drv0.o
C:/sgdk/bin/sjasm.exe -isrc -iinc -ires -iC:/sgdk/src -iC:/sgdk/inc src/z80_drv1.s80 src/z80_drv1.o80 out.lst
SjASM Z80 Assembler v0.39g6 - www.xl2s.tk
Pass 1 complete (0 errors)
Pass 2 complete
Errors: 0
C:/sgdk/bin/bintos.exe src/z80_drv1.o80
C:/sgdk/bin/mkdir.exe -p out/src/
C:/sgdk/bin/gcc.exe -x assembler-with-cpp -m68000 -Wall -Wextra -Wno-shift-negative-value -Wno-main -Wno-unused-parameter -fno-builtin -Iinc -Isrc -Ires -IC:/sgdk/inc -IC:/sgdk/res -BC:/sgdk/bin -O3 -fuse-linker-plugin -fno-web -fno-gcse -fno-unit-at-a-time -fomit-frame-pointer -flto -MMD -c src/z80_drv1.s -o out/src/z80_drv1.o
C:/sgdk/bin/sjasm.exe -isrc -iinc -ires -iC:/sgdk/src -iC:/sgdk/inc src/z80_drv2.s80 src/z80_drv2.o80 out.lst
SjASM Z80 Assembler v0.39g6 - www.xl2s.tk
Pass 1 complete (0 errors)
Pass 2 complete
Errors: 0
C:/sgdk/bin/bintos.exe src/z80_drv2.o80
C:/sgdk/bin/mkdir.exe -p out/src/
C:/sgdk/bin/gcc.exe -x assembler-with-cpp -m68000 -Wall -Wextra -Wno-shift-negative-value -Wno-main -Wno-unused-parameter -fno-builtin -Iinc -Isrc -Ires -IC:/sgdk/inc -IC:/sgdk/res -BC:/sgdk/bin -O3 -fuse-linker-plugin -fno-web -fno-gcse -fno-unit-at-a-time -fomit-frame-pointer -flto -MMD -c src/z80_drv2.s -o out/src/z80_drv2.o
C:/sgdk/bin/sjasm.exe -isrc -iinc -ires -iC:/sgdk/src -iC:/sgdk/inc src/z80_drv3.s80 src/z80_drv3.o80 out.lst
SjASM Z80 Assembler v0.39g6 - www.xl2s.tk
Pass 1 complete (0 errors)
Pass 2 complete
Errors: 0
C:/sgdk/bin/bintos.exe src/z80_drv3.o80
C:/sgdk/bin/mkdir.exe -p out/src/
C:/sgdk/bin/gcc.exe -x assembler-with-cpp -m68000 -Wall -Wextra -Wno-shift-negative-value -Wno-main -Wno-unused-parameter -fno-builtin -Iinc -Isrc -Ires -IC:/sgdk/inc -IC:/sgdk/res -BC:/sgdk/bin -O3 -fuse-linker-plugin -fno-web -fno-gcse -fno-unit-at-a-time -fomit-frame-pointer -flto -MMD -c src/z80_drv3.s -o out/src/z80_drv3.o
[... 101 lines cut ...]
C:/sgdk/bin/mkdir.exe -p out/src/
C:/sgdk/bin/gcc.exe -m68000 -Wall -Wextra -Wno-shift-negative-value -Wno-main -Wno-unused-parameter -fno-builtin -Iinc -Isrc -Ires -IC:/sgdk/inc -IC:/sgdk/res -BC:/sgdk/bin -O3 -fuse-linker-plugin -fno-web -fno-gcse -fno-unit-at-a-time -fomit-frame-pointer -flto -MMD -c src/ym2612.c -o out/src/ym2612.o
C:/sgdk/bin/mkdir.exe -p out/src/
C:/sgdk/bin/gcc.exe -m68000 -Wall -Wextra -Wno-shift-negative-value -Wno-main -Wno-unused-parameter -fno-builtin -Iinc -Isrc -Ires -IC:/sgdk/inc -IC:/sgdk/res -BC:/sgdk/bin -O3 -fuse-linker-plugin -fno-web -fno-gcse -fno-unit-at-a-time -fomit-frame-pointer -flto -MMD -c src/z80_ctrl.c -o out/src/z80_ctrl.o
echo "out/res/libres.o out/src/z80_drv0.o out/src/z80_drv1.o out/src/z80_drv2.o out/src/z80_drv3.o out/src/z80_xgm.o out/src/bmp_a.o out/src/dma_a.o out/src/kdebug.o out/src/maths3D_a.o out/src/memory_a.o out/src/smp_null.o out/src/smp_null_pcm.o out/src/sram_a.o out/src/sys_a.o out/src/tools_a.o out/src/vdp_tile_a.o out/src/bmp.o out/src/dma.o out/src/everdrive.o out/src/fat16.o out/src/joy.o out/src/map.o out/src/mapper.o out/src/maths.o out/src/maths3D.o out/src/memory.o out/src/pal.o out/src/psg.o out/src/sound.o out/src/sprite_eng.o out/src/sram.o out/src/string.o out/src/sys.o out/src/tab_cnv.o out/src/tab_log10.o out/src/tab_log2.o out/src/tab_sin.o out/src/tab_sqrt.o out/src/tab_vol.o out/src/timer.o out/src/tools.o out/src/types.o out/src/vdp.o out/src/vdp_bg.o out/src/vdp_dma.o out/src/vdp_pal.o out/src/vdp_spr.o out/src/vdp_tile.o out/src/vram.o out/src/xgm.o out/src/ym2612.o out/src/z80_ctrl.o" > out/cmd_
C:/sgdk/bin/gcc.exe -BC:/sgdk/bin -n -T C:/sgdk/md.ld -nostdlib out/sega.o @out/cmd_ C:/sgdk/lib/libmd.a C:/sgdk/lib/libgcc.a -o out/rom.out -Wl,--gc-sections
C:\Users\N\AppData\Local\Temp\cckMG2lM.ltrans1.ltrans.o: In function `_reset_entry':
<artificial>:(.text+0xb08): undefined reference to `main'
C:\Users\N\AppData\Local\Temp\cckMG2lM.ltrans1.ltrans.o: In function `_start_entry':
<artificial>:(.text+0xbfa): undefined reference to `main'
<artificial>:(.text+0xcd8): undefined reference to `main'
make: *** [out/rom.out] Error 1
rm src/z80_xgm.s src/z80_drv0.s src/z80_drv1.s src/z80_drv3.o80 res/libres.rs src/z80_xgm.o80 src/z80_drv2.o80 src/z80_drv2.s src/z80_drv0.o80 src/z80_drv3.s src/z80_drv1.o80`

There is an error at the end, not sure if this is ok or not.

If I am creating a new project and including genesis.h, the IDE complains about not finding "libres.h". And indeed, there is no libres.h to be found in sgdk\inc, where genesis.h is located. Does "libres.h" have to be in sgdk\inc?

Is this maybe a known issue? I saw this:
https://github.com/Stephane-D/SGDK/issues/141
But it was closed.

I tried recompiling several times, but the result was the same.

Any help is appreciated.

dantel35, 2022-02-05:
Hm, despite the compiling error, it seems to work now.

In case anyone else has trouble with this to work in VSCODE,
I had to add:
${default}
${GDK}/inc
${GDK}/res

To my include path in VSCode/C/c++ Configurations.
I saw that a libres.h is located in the sgdk\res folder, hence I added it.

One way to do this:
1. Hover with your mouse over the #include <genesis.h> part which is showing the error in VSCode
2. Click on Quick Fix in the upcoming tooltip
3. Click on include "editPath" Setting
4. scroll down until you see the Include path section and add the paths to the field saying "One inclue path per line"
5. Restart VSCode.

After that, it started to work. At least I can run the hello world samples and there are no errors in the IDE.

Still not sure if I have to worry about the compilation errors.

Stephane-D, 2022-02-07:
@dantel35 
The command you used `C:\sgdk\bin\make -f C:\sgdk\makefile.gen` is made to compile your project.
Here you tried to use it from SGDK folder mean that you're compiling *the library itself*, which normally you don't need to do (and if you really want to then use the appropriate makefile : `C:\sgdk\bin\make -f C:\sgdk\makelib.gen`)
If what you wanted is to build your own project (and i guess that is what you want to do), then you can just execute the command `C:\sgdk\bin\make -f C:\sgdk\makefile.gen` **from the project folder** and that's.

I don't know which tutorial you used but a lot of people get confuse with VSCode tutorial they found on internet.
I really suggest to start by just trying to build from command line following the _official instructions_ in the wiki :
https://github.com/Stephane-D/SGDK/wiki/SGDK-Usage

Then when you got it, you may try to integrate the build process into an IDE (as VSCode).

closed 2022-02-07 by Stephane-D