Stephane-D/SGDK #242: SGDK tools

vschwaberow, 2022-02-06:
Just a question: Isn't it giving a benefit for performance and memory management if the additional SGDK tools, esp. rescomp, are rewritten in a modern dialect of C++ or even in Rust? What do you think?

Stephane-D, 2022-02-07:
Rescomp was historically wrote in pure C and to be honest it was a pain to maintain / make it evolve because of that.
I know using Java isn't really fun as you require to install Java JVM to make SDK companion tools to work but it offers 2 big advantages (at least for me):
- I'm much more efficient using Java language than C++ / Rust (i never programmed using Rust)
- Java _binary_ (JAR file) doesn't even need rebuilding whatever is the target platform. C++ / Rust are portable but you still often have to add a bit of specific code to make it _really_ portable which isn't the case in Java.

And finally Java performance by itself is definitely not that bad, it comes really close to C++ / Rust when properly used.
But again for me it's just because Java is my first programming language and it would take me much much more time to develop these tools using C++ for instance.

vschwaberow, 2022-02-07:
I understand that. I would not criticize Java by its performance, but by its memory usage.  Thanks for your statement, Stephane.

Stephane-D, 2022-02-08:
It's true that memory usage is not optimal by any way but just for tooling i think it's not a too big deal. rescomp, lz4w or apj packers should not be very hungry about memory usage (and if they are, it's probably a bug on my side).