Stephane-D/SGDK #249: Can't build docker container.

brandorf, 2022-05-01:
When trying to build the docker container on windows, I get the following error:

```
D:\SGDK>docker build -t sgdk .
[+] Building 0.8s (9/10)
 => [internal] load build definition from Dockerfile                                                               0.0s
 => => transferring dockerfile: 32B                                                                                0.0s
 => [internal] load .dockerignore                                                                                  0.0s
 => => transferring context: 35B                                                                                   0.0s
 => [internal] load metadata for docker.io/amd64/ubuntu:20.04                                                      0.5s
 => [1/6] FROM docker.io/amd64/ubuntu:20.04@sha256:9d42d0e3e57bc067d10a75ee33bdd1a5298e95e5fc3c5d1fce98b455cb8792  0.0s
 => [internal] load build context                                                                                  0.0s
 => => transferring context: 18.99kB                                                                               0.0s
 => CACHED [2/6] RUN dpkg --add-architecture i386   && apt-get update    && DEBIAN_FRONTEND=noninteractive apt-ge  0.0s
 => CACHED [3/6] RUN useradd -ms /bin/sh -d /sgdk sgdk                                                             0.0s
 => CACHED [4/6] COPY . /sgdk                                                                                      0.0s
 => ERROR [5/6] RUN /sgdk/bin/create-bin-wrappers.sh                                                               0.3s
------
 > [5/6] RUN /sgdk/bin/create-bin-wrappers.sh:
#9 0.247 /bin/sh: 1: /sgdk/bin/create-bin-wrappers.sh: not found
------
executor failed running [/bin/sh -c /sgdk/bin/create-bin-wrappers.sh]: exit code: 127

D:\SGDK>
```
The dockerfile looks correct, so I'm not too sure what the issue might be.

Stephane-D, 2022-05-02:
The docker is made for *nix systems only, SGDK is native for Windows so you don't need to use Docker build here.
I guess it fails when trying to execute the #!/bin/bash command on Windows.

dleslie, 2022-05-04:
FWIW, I have been working on the docker container for `marsdev`, which includes the SGDK, and it works fine on Windows and other systems.

You can pull it as `dleslie/marsdev`, it should be PR'd to http://github.com/andwn/marsdev soon.

benjamin-allion, 2022-10-08:
Hello, first thank you very much @Stephane-D for all your work on this SDK 🙏.

About building the docker image from windows, the problem comes from git.
When you're using git (with the default configuration) and trying to clone a git project, it replaces all end of line formats by the `crlf` format (on windows only) :

[image]

And because of that, this error occurs when docker's trying to run `.sh` file :

[image]

To solve that issue, I just added a new `.gitattributes` file to specify to git that it should not change any end of line format for all `.sh` files. 
And now, if any users are trying to clone the project and build the image, no more `CRLF` and so, no more error to generate the image under linux and under windows : 

[image]
[image]

I tried to build the image with that new git configuration on `MacOs` & `Windows` and now everything's fine.

Stephane-D, 2022-10-10:
I just merged your MR ! As i'm using windows, I never tested the docker script myself so thanks a lot for having sorted that issue out !

closed 2022-10-10 by Stephane-D