Stephane-D/SGDK #250: Building project sometimes blows away src/boot/sega.s

brandorf, 2022-05-01:
I'm set up in VS2022 using a makefile project, sometimes a build will modify the src/boot/sega.s file, making it blank.  Future builds will then fail to start in the emulator.

[image]

Since I have the sgdk install location still under source control, I can revert sega.s back to before it was blanked out, but I'm not sure what mechanism is changing that file in the first place.

Stephane-D, 2022-05-02:
I never experienced the problem using make from command line so i guess here is something wrong in the way makefile project are handled in VS2022.
The makefile does copy sega.s file from SGDK origin if not yet present in the project, also it will overwrite it if the SGDK version is more recent. I guess something get wrong in the process but why, i don't know :-/

Stephane-D, 2022-08-22:
Was never able to reproduce it, probably an issue from VS2022 interrupting make script aggressively ?

closed 2022-08-22 by Stephane-D