Stephane-D/SGDK #260: Problem zero size

VitalyDyachkov, 2022-08-04:
Good day, friends! Thanks for the detailed description of the tool and the SGDK tool itself. I'm having a problem compiling a file using the command line. Output ->

C:\SegaProject>C:\SGDK\bin\make -f C:\SGDK\makefile.gen
C:/SGDK/bin/mkdir.exe -p src/boot
C:/SGDK/bin/mkdir.exe -p out
C:/SGDK/bin/cp.exe C:/SGDK/src/boot/sega.s src/boot/sega.s
C:/SGDK/bin/cp.exe C:/SGDK/src/boot/rom_head.c src/boot/rom_head.c
C:/SGDK/bin/gcc.exe  -DSGDK_GCC -m68000 -Wall -Wextra -Wno-shift-negative-value -Wno-main -Wno-unused-parameter -fno-builtin -fms-extensions -Iinc -Isrc -Ires -IC:/SGDK/inc -IC:/SGDK/res -BC:/SGDK/bin -O3 -fuse-linker-plugin -fno-web -fno-gcse -fno-unit-at-a-time -fomit-frame-pointer -flto -c src/boot/rom_head.c -o out/rom_head.o
C:/SGDK/bin/objcopy.exe -O binary out/rom_head.o out/rom_head.bin
C:/SGDK/bin/gcc.exe -x assembler-with-cpp -Wa,--register-prefix-optional,--bitwise-or  -DSGDK_GCC -m68000 -Wall -Wextra -Wno-shift-negative-value -Wno-main -Wno-unused-parameter -fno-builtin -fms-extensions -Iinc -Isrc -Ires -IC:/SGDK/inc -IC:/SGDK/res -BC:/SGDK/bin -O3 -fuse-linker-plugin -fno-web -fno-gcse -fno-unit-at-a-time -fomit-frame-pointer -flto -c src/boot/sega.s -o out/sega.o
src/boot/sega.s: Assembler messages:
src/boot/sega.s:52: Error: skip (0) or count (256) invalid for file size (0)
make: *** [out/sega.o] Error 1

D0NM, 2022-08-05:
Well  the problem is:
rom_head.bin is not generated. IT is basically 0 bytes.

The ugly fix (lol) 

segas.s
line 52

it uses directive .**incbin "file"[,skip[,count]]**
so... replace the zero-length file with anything.
And it'd work. 

Also it is strange, but SGDK has 2 couples of sega.s + rom_head.c
one is licated in sdgk\src\boot
the other is in the template

VitalyDyachkov, 2022-08-05:
Thanks for the answer! But I just didn't understand why the .bin file wasn't generated.

D0NM, 2022-08-05:
The make file has been changed. There is a bug.

VitalyDyachkov, 2022-08-05:
That is, you need to roll back to an earlier version?

VitalyDyachkov, 2022-08-05:
SGDK 1.70 (February 2022)
* MAKEFILE
 - added 'EXTRA_FLAGS' entry to makefiles so we can use custom extra compilation flags

 Is that what you mean?)

VitalyDyachkov, 2022-08-05:
Thank you very much, D0NM friend! Downloaded SDK version 1.34 (December 2017), managed to build a starter project!

D0NM, 2022-08-08:
**My future investigation shows** that road_head.c is being compiled.
But the object file is ~ 4K instead of 988 bytes. Also it has some garbage inside, You cannot see the real text strings in this .o file.

On using objcopy.exe it produces the zero length .bin file

Stephane-D, 2022-08-09:
Fixed ! (looks like -O1 flag break rom_head.c compilation, reverted to DEFAULT_FLAGS)

closed 2022-08-09 by Stephane-D