Stephane-D/SGDK #262: Warning: Don't work with 'Stefan Hyltoft', this man is a faker, bad guy who didn't pay after employed my mate Phoon.

cerberus-master, 2022-08-20:
My mate Phoon contracted with Stefan on the 'Image-Recognition' React Native Project.
He worked whole week around Stefan's requirement. But after all delivered result, he refused to pay without any reason.

I am not posting this issue to get money from him, I hope no one will be deceived by him.

https://user-images.githubusercontent.com/101217159/185763388-87e43453-17bb-4173-b776-181689cd79e6.png
https://user-images.githubusercontent.com/101217159/185763374-16db443a-dbb1-4d39-8030-024bf7c9c4ea.png
https://user-images.githubusercontent.com/101217159/185763364-753a1f07-bf1a-4763-a0cd-0d6da6de70de.png
https://user-images.githubusercontent.com/101217159/185763380-74b5a776-e5d2-4031-af96-cefae09d8667.png
https://user-images.githubusercontent.com/101217159/185763306-5bd417b7-6f39-4a3c-b489-85780c3cbf86.png

matteusbeus, 2022-08-20:
This forum is for posting issues with the SGDK. Not for airing dirty laundry regarding sour partnerships.

Astro1218, 2022-08-20:
I know, but we can't work convenient with this kind of people.
After clean up your place, we always get on the work, didn't we?