Stephane-D/SGDK #264: The current revision can't compile images that use more than 1 pallete

saintienn, 2022-08-29:
I believe that some changes to adapt resources to tmx files, cause some strange side effects when read images
In the commit 84ce6bef38de7e2aa9b3cd748ebef29d034d1676 all my resources that use two palletes run perfectly, but now is showing this error: 
.\||Error: pixel at [408,399] reference a different priority (1 != 0).|
.\||Error: cannot compile resource 'TILESET babaplace_foreground_tileset "stage/babaplace/fgprio.png" APLIB ALL'|
.\||Error: pixel at [408,399] reference a different priority (1 != 0).|
 res\stage.rs||No such file or directory|
.\||warning: '-x assembler-with-cpp' after last input file has no effect|
.\||error: no input files|

Stephane-D, 2022-08-29:
Hi @saintienn 

Is is possible to share you image ? It seems a certain pixel is using different priority (color index >= 128) in a same tile.
Recently i fixed a bug preventing transparent pixel to store priority information.

saintienn, 2022-08-29:
yes, thank you so much! 
In this image I paint some tiles with priority
https://www.dropbox.com/s/x561dsaywte1sju/fgprio.png?dl=0

Stephane-D, 2022-08-30:
So indeed it's related to the bug i fixed.
You have a tile which share pixels from low priority palette (the transparent pixels) and pixels from high priority palette : 
[image]

As reported by the error message : Error: pixel at [408,399] reference a different priority (1 != 0)
It's important than all pixels from a tile reference the same palette (even transparent pixels) :)
Before transparent pixels could be from any palette but that was causing issue when we wanted use transparent but high priority tile (for hilight / shadow effect for instance) so I had to fix that.

closed 2022-08-30 by Stephane-D

saintienn, 2022-08-30:
I fixed the image how you showing, and this tileset come back to compile correctly! 
Thanx Stephane!