Stephane-D/SGDK #274: VDP_setTileMapEx WINDOW not functional

greatAznur, 2022-10-17:
hi  i test ..  the sgdk  and in doc  of  function VDP_setTileMapEx
 the plane can be window but i try  with "BG_A" of "BG_B"  the tilmap as displayed
 but with "WINDOW" plane nothing 
 
VDP_setTileMapEx(BG_B, ctmap, TILE_ATTR_FULL(PAL1, FALSE, FALSE, FALSE, idx2), 0, 0, 0, 0, 20, 28, CPU); functional

VDP_setTileMapEx(WINDOW, ctmap, TILE_ATTR_FULL(PAL1, FALSE, FALSE, FALSE, idx2), 0, 0, 0, 0, 20, 28, CPU);  not diplayed

i change just  the first  argument   ? i forgot something ?

Stephane-D, 2022-10-17:
You need to enable the window first using the VDP_setWindowHPos() and VDP_setWindowVPos() method

closed 2022-10-17 by Stephane-D

greatAznur, 2022-10-17:
oh i have thinks that use VDP_setWindowHPos()  but ... after not before

Stephane-D, 2022-10-17:
Yeah it's normal, BG_A and window are shared so they can't display on same area.

greatAznur, 2022-10-17:
ok thank for  precisions