Stephane-D/SGDK #278: VDP_loadFont() omits duplicate chars

moon-watcher, 2022-11-20:
My custom font use the same char for 0 and O and here is the result:
[image]

It doesn't load the O since 0 is already loaded

Stephane-D, 2022-11-21:
Did you disabled optimisation when defining the font resource (probably as TILESET) ? otherwise rescomp will remove any duplicate tiles in the tileset.

moon-watcher, 2022-11-21:
My fault. Had this line on res file
`TILESET font01 "gfx/font/font.png" NONE`

Since it's a font I assumed erroneous things and missed 4th parameter for optimization.

May be ok to add a new resource FONT just like TILESET but without tile optimization

closed 2022-11-21 by moon-watcher