Stephane-D/SGDK #284: Fail to build on windows

skarab, 2023-01-15:
Hello, I just got the latest version and when I build I get this :

src/boot/sega.s:256: Error: invalid operands (*UND* and *UND* sections) for `+'

skarab, 2023-01-15:
Ok my makefile is outdated, sorry

closed 2023-01-15 by skarab

Stephane-D, 2023-01-16:
Yeah It was, I recently added some options in build command to accept these caracters :)

skarab, 2023-01-16:
Yes, gonna upgrade to latest rev, it's a huge step up ! (was using version just before the whole sprite system rewrite :)

Strugglemeat, 2023-01-24:
I'm having the same issue, where can I get a new makefile?

Stephane-D, 2023-01-24:
Just directly download the last SGDK github archive, that should work :)

skarab, 2023-01-24:
New makefile is in SGDK/makefile.gen (I had copied it to personnalize in old version, my trouble was I forget to update it)