Stephane-D/SGDK #296: SpriteEngine - auto anim

skarab, 2023-09-03:
Hi! By now it's not possible to bypass the auto frames animation, it looks its reset in the visibility update, I have hard patched it in the lib but it would be cool to have an option to wholy disable sprite->timer things.
Cheers!

Stephane-D, 2024-04-03:
Not sure to understand this one, normally setting sprite->timer to 0 is enough to disable auto animation.

skarab, 2024-04-11:
Great thanks.

closed 2024-04-11 by skarab

skarab, 2024-04-21:
Just a side note, to get it working we need to set sprite->timer to 0 at each frame, else it got overwriden by static u16 updateFrame(Sprite* sprite, u16 status)

Stephane-D, 2024-04-24:
Indeed, it requires to reset timer to 0 each time you change frame.
I added a new SPR_setAutoAnimation(sprite, bool) method to take care of that in a better way :)