Stephane-D/SGDK #298: Rescomp freeze for some sprite resource in compilation time

saintienn, 2023-11-09:
Its very strange but happen for me. 
I believe it's a bug in the new sprite crop improvement

SPRITE spr_goku_spn0 "sprite/squatpunch01/p00.png" 10 10 FAST 0 NONE TILE FAST
[image]

SPRITE spr_goku_ck1 "sprite/chargeki/ck01.png" 10 12 FAST 0 NONE TILE FAST
[image]

Stephane-D, 2023-11-09:
Thanks for reporting, indeed this is due to some (stupids) recent changes, sorry for that !
It's fixed now. Also i suggest you to use "BALANCED MEDIUM" as last parameters for your sprites as "TILE FAST" really gives too many sprites (13 and 16 sprites versus 8 and 10 sprites). The tile number increase a bit but worth the gain on the number of hardware sprites imo.

closed 2023-11-09 by Stephane-D

saintienn, 2023-11-10:
Thanks!! I will follow your tip