Stephane-D/SGDK #299: Optimization idea for rescomp

thomasgoldstein, 2023-11-15:
Hello.

I wanted to suggest the following optimization for rescomp, which would make our life much easier:

Add the ability (as default behavior or as an optional flag) to skip the generation of the output files (*.d and *.o) if their last modified date is more recent than the last modified date of the source png image.

That would make our game's compilation process much faster and would allow us to use the slower but more optimal compression levels of rescomp.

What do you think? Thank you.

thomasgoldstein, 2023-11-15:
I'm closing this because I think it's not a useful suggestion after all, and I'm going to consider alternatives. Sorry!

closed 2023-11-15 by thomasgoldstein

Stephane-D, 2023-11-15:
Hi @thomasgoldstein 

Normally if make was doing a correct job, it would only rebuild required file and so using rescomp to rebuild resources only if resource files as .png were modified, the .d file are generated for that exact reason (to properly handling file dependency). The problem is that make is not handling that correctly and usually rebuild the whole even when they didn't changed. I should investigate that problem, maybe the issue is in the makefile itself.
Still you solution is probably better suited for your needs ;)