Stephane-D/SGDK #300: Rescomp could detect duplicate frames in case of same sprite sheet with different colors

thomasgoldstein, 2023-11-18:
Hi.

Consider the following scenario: 2 character sprite sheets that contain the exact same character frames, just with different colors (color indexes). In that case, rescomp will not realize it could avoid performing the sprite cutting on the second sheet.

I believe this is a relatively common scenario to work around the harsh color palette limitations of the Mega Drive. Xeno Crisis does this for the green and blue variants of the playable characters, for example.

What I guess you're doing internally in rescomp's sprite cutting logic is only care about which pixel is opaque or transparent, you don't care about the colors used, right? That means you could use that opaque vs transparent pixel matrix as a reference to determine whether a frame is a clone of an already-processed frame or not, which would cover the need above.

Just an idea. Thanks for entertaining the idea!

PS: This also makes me wonder if, ROM data wise, any data could be shared across such similar sprite sheets in order to reduce the footprint of similar sheets.

thomasgoldstein, 2023-11-18:
Here are example sprite sheets for your convenience.
Rescomp commands:
```
SPRITE sprite "sonic_blue.png" 11 11 NONE 0 NONE TILE
SPRITE sprite "sonic_red.png" 11 11 NONE 0 NONE TILE
```

[image]
[](url)

[image]

Stephane-D, 2023-11-23:
Hi @thomasgoldstein 

Thanks the detailed report and the sprite sheets !
Indeed rescomp doesn't perform any check about duplicated sprite _mask_ (opaque pixels shape) along all the sprite frames / animations / sheets and it would avoid recomputing the sprite cutting in that case as you said. Still that is not that easy to compare sprite mask along all already processed sprites, i guess I could compute a hash code based on the mask shape instead of the content... 
I qualified your issue an enhancement and keep it open until i find sometime to work on that !

Stephane-D, 2025-04-02:
Just to add a note, the last version of rescomp does actually detects duplicated sprite mask but unfortunately for now it does it only for a single animation row, it would require more work to allow doing that on a complete spritesheet or even in the whole sprite base.