Stephane-D/SGDK #302: SPR_getAnimationDone is delayed

RodrigoMedeirosRS, 2023-11-30:
Hello, when using the SPR_getAnimationDone function I identified that it has a Delay, basically it is considering that an animation ends when it returns to the first frame and not when it reaches the last, which can cause flickering in the animations.

Sample 1:
- Pseudo Code:` if(Sprite.frameInd == Sprite->animation->numFrame-1) { change animation; }`
- Result (No Delay, no flickering) https://youtu.be/r7S18oDZAGo

Sample 2:
- Pseudo Code:` if(SPR_getAnimationDone(Sprite) stop animation. { change animation; }`
- Result (With Delay, with flickering) https://youtu.be/eVFn4gym_BE

D0NM, 2023-11-30:
It is not a bug. 
Why? ))) is's obvious

D0NM, 2023-11-30:
in short.
it is not a bug.
it's normal

bool SPR_getAnimationDone(Sprite* sprite)
{
    return (sprite->status & STATE_LOOPED)?TRUE:FALSE;
}

it returns TRUE when the loop happens.
I use it in our games a lot.

D0NM, 2023-11-30:
also in the code i do not use it to stop the animation.
i use my own functions for that.

this function is used to know is the animation ended.

RodrigoMedeirosRS, 2023-11-30:
The problem is that the function is not returning when the animation done, but rather when it restarts, which causes flickering.

Notice in the video when I use the function, the animation is displayed in full, and the change only occurs after the first frame is repeated.

The problem is that you are not identifying the end of the animation but its beginning.

RodrigoMedeirosRS, 2023-12-01:
Thanks for the tip! Everthing is running fine now :)

closed 2023-12-01 by RodrigoMedeirosRS

Stephane-D, 2023-12-01:
Just added SPR_setAnimationLoop(..) method to disable auto animation loop.
So using that with SPR_getAnimationDone(..) is enough to get the sprite properly handled when animation is done avoiding potential first frame display glitch

D0NM, 2023-12-02:
I checked the source, 
so, now, when the animation with DISABLED LOOPING is done, 
you do not trigger any **callback** on the frame change.
Am I right?

Stephane-D, 2023-12-02:
Basically yeah as it stays on the last frame but I completely understand the problem with that.. I will make a change to let the frameChangeCallback still happen on animationDone change state:-)

Stephane-D, 2023-12-03:
> I checked the source, so, now, when the animation with DISABLED LOOPING is done, you do not trigger any **callback** on the frame change. Am I right?

I modified the code so you will receive a last frame change call on animation end even when animation loop is disabled, necessary to get the ANIMATION_DONE state in the callback even if the frame didn't really changed.