Stephane-D/SGDK #303: SGDK is not running on Mega Drive / Sega Genesis

RodrigoMedeirosRS, 2023-12-14:
When compiling the code with the Release version of SGDK and writing it to a cartridge, everything works. However, when writing the same code in the Master branch version, the video game cannot run the rom.

In the emulator everything works normally, tested in Kega Funsion and KGens.

Tested hardware:
- Mega Drive 3 - Tectoy with Sega CD input
Observation:
- Mega Drive 3 - Tectoy is equivalent to the American Sega Genesis 2

Stephane-D, 2023-12-14:
Both Kega nor Gens are accurate emulators, Better to use BlastEm.
About the issue, without more information it's impossible to sort out the problem as it's all working for me (SGDK samples does work).
Did you properly rebuilt the library with the last sources then did a full rebuild of your project ?

RodrigoMedeirosRS, 2023-12-14:
I recompiled the lib and the entire project, I even have some build scripts that delete the outputs from the out folder, ensuring that the solution is always recompiled.

I did some tests earlier, I noticed that part of the problem occurred when I compiled from MARSDEV on my Linux machine, compiling from a Windows machine, using only the SGDK, what happens is the following.

The game does not load, the entire screen is stuck showing only the first tile of BG_A. I'm going to try to redo the tests with Blastem to see if I can catch any logs or errors that will help shed some light, in the other emulators there are no errors.

Note: When I discovered the error, I returned my SGDK to version 1.90, took the changes from the commit, and isolated only what related to the SPR_getAnimationDone and SPR_setAnimationLoop functions and applied them to the release 1.90 source. After recompiling the lib and the game, everything started working again (at least generating the build on Windows).

I'll try to do a battery of tests and see if I find anything.

RodrigoMedeirosRS, 2023-12-14:
No Error's in BlastEm, everthing working from every build made both in Windows and Linux machine, with SGDK from main.
I will try another test pulling all corrections from the commit into 1.90 branch.

Stephane-D, 2023-12-15:
> No Error's in BlastEm, everthing working from every build made both in Windows and Linux machine, with SGDK from main. I will try another test pulling all corrections from the commit into 1.90 branch.

Do you mean it works correctly from last changes in "main" branch ?
At least I can't find any issues when building and testing all samples from main branch in my case

closed 2024-01-31 by Stephane-D