Stephane-D/SGDK #305: Question about bank switching

Oliver-makes-code, 2024-01-14:
Sorry if this isn't the right place to ask, I couldn't find an alternative place.
Does this automatically determine what banks functions should go into, and automatically switch banks depending on which function needs to be run?
If not, is there a way to tell it to put functions into certain banks?
Thanks for your help!

skarab, 2024-04-11:
Curious as well. I think it has been removed in latest version ? A bit hard to stick on 4MB for my game, but ive read 8MB is ok without bank switching, is it correct ?

skarab, 2024-04-11:
(but 8MB with save slots cartdrige would be hard to find i guess)

Stephane-D, 2024-04-24:
SGDK does include a kind of auto bank switch mechanism but it need to be enabled through the SGDK inc/config.h file and then you need to rebuild the library. 
As mentioned there is a way to do 8 MB without bank switch and with very few logic (hardware wise), at the price of not being able to use the MCD using that configuration.

closed 2024-09-03 by Stephane-D