Stephane-D/SGDK #307: How can we cut sprites?

thomasgoldstein, 2024-01-27:
Hello!

I'd like to be able to only render part of a sprite. For example, just displaying the top of a character's sprite, and omit the bottom 16px (or whatever). From what I understand, the Mega Drive cannot do this natively.

But it has been done in the Punisher: https://youtu.be/Cje2UHXIa24?t=2698

How did they do it?
Thinking about it, I believe it's like they basically have 2 sprite cutting definitions to be applied on the same sprite sheet graphics.
Since rescomp automatically cuts sprite frames for you, I guess such things won't be possible unless rescomp lets you manually define how frames should be cut, and on top of that, lets you define multiple ways to cut sprite frames. Does that sound correct?

I understand this may be out of the scope of the SGDK or rescomp, because it may be too low level (as it might kinda break the current design logic), but I just wanted to confirm it.

Thank you.

thomasgoldstein, 2025-03-12:
Hi @Stephane-D, should this issue be closed, or do you have ideas to support this at some point?

Stephane-D, 2025-03-13:
> Hello!
>
> I'd like to be able to only render part of a sprite. For example, just displaying the top of a character's sprite, and omit the bottom 16px (or whatever). From what I understand, the Mega [...]

Hi @thomasgoldstein 

Forgot to reply you here but yeah I think that is not something rescomp can easily manage, the automatic cutting process is already complex by itself, adding that sort of constraint above it seems impractical.
The usual solution is to split your sprite in 2 parts so you can independently animate bottom and top part of the body (Haggar character in Final Fight uses the same configuration, only the bottom of the body is animated when walking while the top is fixed for instance).
So yeah we can probably close the issue by itself, it's more an engine design thing :)