Stephane-D/SGDK #309: Problems with XGM_StartPlayPCM function

saintienn, 2024-02-01:
Hello Stef!

I updated SGDK just now and some weird things happened.
When I compile I get an error like
"error: this method is obsolete, use SOUND_PCM_CH4"
for this line: XGM_startPlayPCM(73, 10, SOUND_PCM_CH4);

But when I change SOUND_PCM_CH4 to number 4, it compiles but...
When I play this several times the background sound stops.

if I change SOUND_PCM_CH4 to SOUND_PCM_CH_AUTO no sound will be played (only the background music)

I use the XGM_startPlay function for the vgms and the XGM_startPlayPCM function for the WAV effects

saintienn, 2024-02-01:
May be some issue related with CH1 PCM, when the background stop  if I excute XGM_resumePlay(); it play again

Stephane-D, 2024-02-01:
Hello !

Are you sure you're on the last version ? because SOUND_PCM_CH4 shouldn't produce any error (Z80_DRV_CH3_SFT does).
Also SOUND_PCM_CH_AUTO isn't supported by XGM driver (see the documentation of XGM_startPlayPCM(..) method).
And SOUND_PCM_CH1 is normally reserved for background music so better to not use it for SFX !

saintienn, 2024-02-02:
Rigjt, I Will try again cloning in a clean directory, but I'm only pull and after recompile the sgdk

saintienn, 2024-02-03:
Well, I cloned it to a clean directory and compiled it and All Its Works! thanx Stef <3

closed 2024-02-03 by saintienn