Stephane-D/SGDK #326: Wrong JOY_getJoypadType(JOY_2) result

D0NM, 2024-04-18:
What I have (in BlastEm)

PORT1:  PAD6
PORT2: TEAMPLAYER
in TEAMPLAYER's PORT1:  PAD6
(ofc, I enable TEAMPLAYER support on PORT2 manually)

Both JOY_1 and JOY_2 work fine.  The data reading, the callbacks work fine.

ERROR:
JOY_getJoypadType(JOY_2) returns  JOY_TYPE_UNKNOWN  when it should return JOY_TYPE_PAD6

I suppose it might be a bug for all gamepads connected via TEAMPLAYER or EA 4-WAY PLAYER.

Peace!

D0NM, 2024-04-18:
i checked your code.... maybe it is all hw limitation...
or bad emulation.

i just want to count the number of connected gamepads.. and i cannot do it

Stephane-D, 2024-04-24:
It would be nice to confirm the issue on real hardware. The code was wrote a long time ago by Chilly Willy and as far i remember he validated against real hardware. Still maybe some devices has not been fully tested.

RealMalachi, 2024-05-09:
If it worked before, it could be a regression. If I had to guess, maybe ram isn't being properly allocated, making it bleed into other controller ram data?
I own a multitap I can test this with; team player with "ExtrA" mode. Might test SGDKs implementation later, have other controllers I need to test anyway

D0NM, 2024-05-09:
Would be cool ti SGDK release had updated sample joy-test rom.bin
It shows current gamepad activity and the detected names of the devices on the screen.

At my side... I added a hacky solution - i listen to all JOY inputs and mark those gamepads which use MODE, X, Y or Z buttons at least once. This hacky solution let me use at least 4 inputs

loki666, 2024-08-02:
Mmmmh that issue does rings a bell to me.

see my joy -> player mapping in my code

```
    if (JOY_getPortType(PORT_1) == PORT_TYPE_EA4WAYPLAY || JOY_getPortType(PORT_1) == PORT_TYPE_TEAMPLAYER) {
        portMapping[0] = JOY_1;
        portMapping[1] = JOY_3;
        portMapping[2] = JOY_4;
        portMapping[3] = JOY_5;        
    } else if (JOY_getPortType(PORT_2) == PORT_TYPE_TEAMPLAYER) {
        portMapping[0] = JOY_2;
        portMapping[1] = JOY_6;
        portMapping[2] = JOY_7;
        portMapping[3] = JOY_8;        
    } else {
        portMapping[0] = JOY_1;
        portMapping[1] = JOY_2;
        portMapping[2] = JOY_7;
        portMapping[3] = JOY_8;
    }
```