Stephane-D/SGDK #338: Input Handling Issue with SGDK on Windows 11

iz2rpn, 2024-06-24:
Sure, here’s the updated issue description with the additional information:

Title: Input Handling Issue with SGDK on Windows 11

Description:

I am testing a project on Windows 11 that uses SGDK to compile a controller pad for the Sega Genesis. The ROM runs, but the keyboard inputs are not being recognized. The only way to detect a pad button press is to perform a hard reset and press a button, which gets recognized only once. Here is the code I am using:

#include <genesis.h>

// Function declarations
static void handleInput();
static void debugInput(u16 value);

// Global variable for the frame counter
static u32 frameCounter = 0;

int main()
{
    // Initialize SGDK
    VDP_drawText("Initializing...", 6, 2);
    JOY_init();
    VDP_drawText("SGDK Initialized", 6, 4);

    while (1)
    {
        // Increment the counter
        frameCounter++;

        // Display the counter
        char str[20];
        sprintf(str, "Frame: %lu", frameCounter);
        VDP_drawText(str, 2, 24);

        // Read controls, called once per frame
        handleInput();

        VDP_waitVSync();
    }

    return (0);
}

static void handleInput()
{
    // Variable to store the pad input
    u16 value = JOY_readJoypad(JOY_1);

    // Debug: show the input value
    debugInput(value);

    // Only if the pad state has changed
    if (value != 0)
    {
        // Clear the text area before drawing new text
        VDP_clearTextArea(0, 7, 40, 28);

        // If left is pressed...
        if (value & BUTTON_LEFT)
            VDP_drawText("LEFT", 8, 10);

        // If right is pressed...
        if (value & BUTTON_RIGHT)
            VDP_drawText("RIGHT", 20, 10);

        // If up is pressed...
        if (value & BUTTON_UP)
            VDP_drawText("UP", 16, 8);

        // If down is pressed...
        if (value & BUTTON_DOWN)
            VDP_drawText("DOWN", 16, 12);

        // If A is pressed...
        if (value & BUTTON_A)
            VDP_drawText("BUTTON A", 6, 18);

        // If B is pressed...
        if (value & BUTTON_B)
            VDP_drawText("BUTTON B", 14, 18);

        // If C is pressed...
        if (value & BUTTON_C)
            VDP_drawText("BUTTON C", 22, 18);

        // If X is pressed...
        if (value & BUTTON_X)
            VDP_drawText("BUTTON X", 6, 20);

        // If Y is pressed...
        if (value & BUTTON_Y)
            VDP_drawText("BUTTON Y", 14, 20);

        // If Z is pressed...
        if (value & BUTTON_Z)
            VDP_drawText("BUTTON Z", 22, 20);

        // If START is pressed...
        if (value & BUTTON_START)
            VDP_drawText("START", 10, 16);

        // If MODE is pressed...
        if (value & BUTTON_MODE)
            VDP_drawText("MODE", 20, 16);
    }
}

// Function to show the input value for debugging
static void debugInput(u16 value)
{
    char str[20];
    sprintf(str, "Input: %04X", value);
    VDP_drawText(str, 2, 26);
}

Stephane-D, 2024-06-25:
You need to call SYS_doVBlankProcess() instead of VDP_waitVSync().
I guess you used an outdated tutorial.
Better to start out from available samples (sample folder) to avoid that kind of issues ;-)

closed 2024-06-25 by Stephane-D

iz2rpn, 2024-06-25:
I'm sorry, I hadn't seen the wiki, I tried it and it works great, it was my emotion, my goal of creating a cartridge is getting closer and closer. thanks for what you do and sorry again