Stephane-D/SGDK #345: XGM2 tracksa plays back at different speeds on PAL and NTSC machines

howprice, 2024-08-16:
`setMusicTempo` in `xgm2.c` earlies out when the driver is loaded, so the tempo is not set. This means that the track plays back at different speeds on PAL and NTSC machines.

I believe this is due to an ordering bug: `Z80_loadDriver()` calls `XGM2_loadDriver`, then sets `currentDriver`. However, `XGM2_loadDriver` calls `setMusicTempo`, which earlies out if `Z80_getLoadedDriver()` (`currentDriver`) is not set, which at this point is is not.

The workaround is to manually call `XGM2_setMusicTempo()` after loading the driver.

thomasgoldstein, 2024-08-16:
It sounds like #340 was not fixed. Good that you have a lead to fix it for real this time.

howprice, 2024-08-16:
Sorry - I forgot to search closed issues. I think the root cause may be the cyclic dependency.

Stephane-D, 2024-09-26:
It looks like my fix was overwritten, the problem is indeed the _setMusicTempo(..)_ which **should not** test for the current driver (unlike XGM2_setMusicTempo(..) which does it) and i'm pretty sure that I already fixed the method before but I probably overwritten it by mistake. Thanks for reporting.

closed 2024-09-26 by Stephane-D

howprice, 2024-09-27:
That's great. Thanks for looking into this.