Stephane-D/SGDK #348: undefined reference to 'main' error while compiling

abadede919302, 2024-08-30:
Hi, I was making the Hello World program in SGDK, and when I compile it i get an error:
```
C:\Users\vella\OneDrive\Desktop\SGDK>set GDK="C:\Users\vella\OneDrive\Desktop\SGDK"

C:\Users\vella\OneDrive\Desktop\SGDK>%GDK%\bin\make -f "C:\Users\vella\OneDrive\Desktop\SGDK\makefile.gen" release && START /B "C:\Users\vella\OneDrive\Desktop\Gens KMod\gens.exe" %CD%/out/rom.bin
C:/Users/vella/OneDrive/Desktop/SGDK/bin/mkdir.exe -p src/boot
C:/Users/vella/OneDrive/Desktop/SGDK/bin/mkdir.exe -p out
C:/Users/vella/OneDrive/Desktop/SGDK/bin/gcc.exe -m68000 -BC:/Users/vella/OneDrive/Desktop/SGDK/bin -n -T C:/Users/vella/OneDrive/Desktop/SGDK/md.ld -nostdlib out/sega.o @out/cmd_ C:/Users/vella/OneDrive/Desktop/SGDK/lib/libmd.a C:/Users/vella/OneDrive/Desktop/SGDK/lib/libgcc.a -o out/rom.out -Wl,--gc-sections -flto
C:\Users\vella\AppData\Local\Temp\cc8MUycR.ltrans5.ltrans.o: In function `_reset_entry':
<artificial>:(.text+0x5b0): undefined reference to `main'
C:\Users\vella\AppData\Local\Temp\cc8MUycR.ltrans5.ltrans.o: In function `_start_entry':
<artificial>:(.text+0x1104): undefined reference to `main'
<artificial>:(.text+0x11da): undefined reference to `main'
make: *** [out/rom.out] Error 1
```
How do i fix this??

thomasgoldstein, 2024-08-30:
The GitHub issues are for bug reports or feature requests, not for support.
For support, you can go to the Discord server: https://discord.gg/xmnBWQS

Stephane-D, 2024-09-03:
Not a bug. (support request)

closed 2024-09-03 by Stephane-D

abadede919302, 2024-09-06:
> The GitHub issues are for bug reports or feature requests, not for support. For support, you can go to the Discord server: https://discord.gg/xmnBWQS

Okay i see. I will use Discord for support next time.