Stephane-D/SGDK #357: PCM audio quality issues

joeyparrish, 2024-10-01:
I'm having trouble with PCM audio playback quality.  I have carefully filtered my PCM audio to avoid aliasing during downsampling, and my downsampled 13kHZ 8-bit PCM stream sounds much better when played with `ffplay` than it does when played through SGDK.

The noise I experience during playback varies by which driver I use and what type of content it is, but it's basically a low static.  For pop music, it's usually less noticeable, but for cinematic content, it seems worse.

I have experimented with the XGM2 and SND_PCM drivers.  Each is noisy in slightly different ways.  SND_PCM tends to be a little "wobbly" in a way XGM2 isn't.  Overall, I like XGM2 better.  I haven't managed to make MDSDRV work yet.  (It plays samples for about 1 second, then just stops.)

 - What is the ideal driver for audio quality if I only care about PCM?
 - Should I consider other drivers than these?
 - How should I debug quality issues?
 - Are there additional filters I should apply to the source audio?
 - What information is necessary for a bug report on audio quality?

I've attached simple source, samples (PCM & WAV formats), and ROMs of playback in different drivers.  I hope it's useful.

[audio-only.zip](https://github.com/user-attachments/files/17213588/audio-only.zip)

 - audio-only/src/main.c: source, with macro to select audio driver
 - audio-only/res/sample.pcm: 13.3kHZ 8-bit signed PCM
 - audio-only/res/sample.res: resource file for PCM samples
 - audio-only/res/sample.wav: WAV version of sample.pcm for easy comparison
 - audio-only/mds.rom: MDSDRV playback ROM (fails)
 - audio-only/pcm.rom: SND_PCM playback ROM
 - audio-only/xgm2.rom: XGM2 playback ROM (best so far)

Thanks for your help!

joeyparrish, 2024-10-02:
> Just set HALT_Z80_ON_DMA and HALT_Z80_ON_IO to 0 in the inc/config.h file then rebuild the library.

I tried that before, but didn't notice much improvement.  I'll try again, though.  It has been a while.

> I can probably try to shorten the Z80 interruption time to improve that though.

Please let me know if this works!  I would be very grateful.

Stephane-D, 2024-10-02:
I just push a change which improve quite a bit the PCM playback with Z80_HALT_ON_IO set to 1.
Don't hesitate to test and report your feeling about it.

joeyparrish, 2024-10-02:
That's a huge improvement.  I can't tell the difference now between Z80_HALT_ON_IO set to 0 and to 1.

Stephane-D, 2024-10-02:
I guess we can close the issue ? :)

joeyparrish, 2024-10-02:
Yes, let's do that.  Thank you for your support!

closed 2024-10-02 by joeyparrish