Stephane-D/SGDK #371: Avoid loading duplicate frames when calling SPR_LoadAllFrames

TheRoboZ, 2024-11-13:
Now that rescomp can recognize duplicated frames in a sprite (not only sequential), when you call SPR_LoadAllFrames, the duplicated frames tiles are loaded multiple times.
Wouldn't it be possible now to just adjust the vdp tile indexes of a frame as to reuse the existing tiles and load it only once? This would save a lot of VRram and also avoid the need for extra code to circumvent the problem. Making sgdk included animation system a lot more flexible.

Stephane-D, 2024-11-21:
SPR_loadAllFrames(..) now detects duplicated tilesets and only load them once in VRAM.
The duplicate check requires more processing and also more RAM to store temporary arrays so the method should perform slower and requires more memory that before. Probably not a big deal though as SPR_loadAllFrames(..) is likely to be called at level loading period :)

closed 2024-11-21 by Stephane-D

TheRoboZ, 2024-11-22:
Thanks, I am testing it and seems the reported totalNumTile still accounts for the original duplicates frames so it leaves gaps when chain-loading many sprites using totalNumTile to increment the index. Is it possible to fix this or will this be a limitation?

As for the possible performance penalty, I haven't had time to check rescomp but, since it detects duplicate frames at compile time, isn't it possible to do as you did for the single sprite optimization and write a special value to avoid runtime duplicate detection again? Like use a negative number to directly point to the other frame tileset or something similar?

Anyway, great to have all the new features